platformio/platformio-vscode-ide #2873: platformio serial monitor config issue

yuanhao9527, 2022-01-11:
I press “Ctrl + T” but it keep show "unknown menu character Ctrl+J" ，I cont config serial at all. 
I can only press "Ctrl +C" to shut down terminal.
please tell me how to solve this problem

[image]

ivankravets, 2022-02-11:
Does it work if you run the device monitor without VSCode?

1. Please open `cmd.exe`
2. `cd %USER_PROFILE%/.platformio`
3. `penv/Scripts/activate`
4. `pio device monitor`

yuanhao9527, 2022-02-19:
thanks, I already know how to config. which I got confused by info:  "meau : ctrl+T", turns out I should press"ctrl+T +meau keys"  to config.  I think you should write the info more clear.

closed 2022-02-19 by yuanhao9527

dzid26, 2023-07-16:
> Does it work if you run the device monitor without VSCode?
> 
> 1. Please open `cmd.exe`
> 2. `cd %USER_PROFILE%/.platformio`
> 3. `penv/Scripts/activate`
> 4. `pio device monitor`

So what to do if this is an actual issue and as you mentioned it works in cmd.exe but not in VSCode terminals?
```
--- Terminal on COM4 | 9600 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Pro--- unknown menu character '\x7f' --
```

XQB12138, 2023-09-22:
> > 如果您在没有 VSCode 的情况下运行设备监视器，它是否有效？
> > 
> > 1. 请打开`cmd.exe`
> > 2. `cd %USER_PROFILE%/.platformio`
> > 3. `penv/Scripts/activate`
> > 4. `pio device monitor`
> 
> 那么，如果这是一个实际问题，并且正如您提到的，它在cmd中有效.exe但在VSCode终端中不起作用，该怎么办？
> 
> 按 Ctrl+T+H：
> 
> ```
> --- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
> unknown menu character '\x7f' --
> ```
> 
> https://www.codetable.net/hex/7f
> 
> Ctrl + T + I但是有效：/

我也遇到这个问题，请问你解决了吗？？

dzid26, 2023-09-22:
I haven't solved it, but when you launch 'pio' in some non-vscode terminal, then help menu shows up correctly:
>cmd.exe
cd %USER_PROFILE%/.platformio
penv/Scripts/activate
pio device monitor

And apart from broken menu display, I think everything else works in vscode.

XQB12138, 2023-10-03:
我也是这样，只有在cmd中运行pio device monitor的时候 能够正常时用，在platformIO中使用串口监视器的时候依然报错
`--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
unknown menu character '\x7f' --`
重装platformIO也不行。

Lulu-Miao, 2024-01-03:
I also encountered the same problem, and my ESP32-CAM development board got stuck after executing the _pio device monitor_.