platformio/platformio-vscode-ide #2881: Start PIO Home Server

JohnCrash, 2022-01-16:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.4.0
  System: Darwin, 15.6.0, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at /Users/john/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:9851
	at ChildProcess.r (/Users/john/.vscode/extensions/platformio.platformio-ide-2.4.0/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at maybeClose (internal/child_process.js:1048:16)
	at Socket.<anonymous> (internal/child_process.js:439:11)
	at Socket.emit (events.js:315:20)
	at Socket.EventEmitter.emit (domain.js:467:12)
	at Pipe.<anonymous> (net.js:673:12)
  ```

Fa-b, 2022-01-24:
I had the exact same error under Windows 10. Removing and reintalling the extension didn't help. What fixed it was removing the `.platformio` folder in the user directory and then reinstalling the extension in VSCode.
What I realised afterwards is that the python version used for Platform IO has changed, before removing the .platformio folder, reinstalling the extension in VSCode had no effect on that, it used a builtin python version 3.7.7 installed at `.platformio\python3`. Now it uses the system wide installed version (version 3.10.2.

I confirmed I receive that error again by changing the python version pointed to in `/Users/xxx/.platformio/penv/pyvenv.conf` back to that python 3.7.7 install (kept a copy of the folder before removing).
So maybe that python install has become corrupt?

I hope this helps you.

Fa-b, 2022-01-24:
@ivankravets sry for directly linking you here, but it seems to me that this is a quite frequent issue and my findings might be of interest to you.

ivankravets, 2022-01-24:
@Fa-b the latest Python in the registry is [3.9.8](https://registry.platformio.org/tools/platformio/python-portable?version=1.30908.0) for windows. Could your remove `~/.platformio/python3` and `~/.platformio/penv` folders. Does it still use your system Python 3.10?

Fa-b, 2022-01-25:
@ivankravets a `~/.platformio/python3` folder was not added again after reinstalling and yes I can confirm that when removing `~/platformio/penv` once again and restarting VSCode, the installation will add it again and link to the systemwide python install:
```
home = C:\Users\Fa-b\AppData\Local\Programs\Python\Python310
include-system-site-packages = false
version = 3.10.2
```

also `python -V` in VSCode Terminal yields 3.10.2.

I wonder if the update to PlatformIO 2.4.0 from within VSCode Extension Manager was supposed to switch paython version to 3.9.8? Before I attempted to fix it manually, I also tried downgrading and again upgrading versions of PlatformIO to no avail.
What I really don't get is, why does this python version suddenly cause issues?