platformio/platformio-vscode-ide #2891: If process unexpectedly quits while debugging, Python will use 100% CPU until force quit

theficus, 2022-01-20:
[summary="Environment summary"]
MacOS 11.6.2
VSCode: 1.63.2
PlatformIO extension: 2.4.0
PlatformIO Core: 5.2.4
[/summary]

When debugging a program in PlatformIO when using the integrated VSCode debugging experience, I've found that if the process unexpectedly quits (for example because of a seg fault), the underlying Python process gets stuck in a tight loop where it uses 100% CPU and doesn't quit until being force killed. This is not visible in VSCode's Process Explorer view.

I originally noticed this when after trying to debug a code issue I found that my laptop fan was going crazy and my laptop battery was draining very quickly.

This is very easy for me to reproduce; if I write some code like this:

```c++
#include "stdio.h"

int main(int argc, char** argv)
{
    char* c;
    *c = 'a';
    return 0;
}
```

Which is quite obviously wrong and will cause a seg fault, once the process exits I can see the Python process using 100% CPU. (It may also be necessary to set a breakpoint before the segfault, but I'm not 100% certain.)

This does not reproduce when using the CLI debugger with `pio`, this only reproduces when using the VSCode integrated debugger.

I'm assuming that the gdb shim to VSCode doesn't gracefully handle the process unexpectedly quitting and gets stuck into a bad state when this happens.

I'm happy to collect more details if needed. Just let me know.

theficus, 2022-01-20:
Discussion on community forum: 
https://community.platformio.org/t/debugger-using-huge-amounts-of-cpu-and-battery-until-python-is-force-quit/25512

theficus, 2022-01-20:
I wonder if this is at all related to #2683 and it picking up the Xcode begin of Python vs the .pio version. It wouldn't be the first time Mac bundled versions of certain tools have different behaviors from mainstream versions. 

In haven't had a chance to try to repro this issue on Windows to confirm this hunch, though.

theficus, 2022-03-04:
Just a quick update -- I'm seeing this when performing both native debugging on my local machine as well as when doing JTAG debugging against a board. It seems to repro easily by simply stopping a debug session while it's in progress (doesn't repro 100% of the time but enough that I feel it necessary to check running processes every time I stop debugging to kill the runaway Python process if needed). This is making debugging a pretty miserable experience. :(

If there's any more information I can provide that may assist in getting this fixed, please let me know.

michikite, 2022-10-11:
Same problem when debugging an esp32 with JTAG.  I am using Python 3.10 using PyEnv. 
Happy to test, if someone has a hint on how to fix this.

britannio, 2024-04-23:
I'm facing this issue too.

McKayRansom, 2024-07-15:
We are running into this issue on the latest MacOS. I will update if we find any steps to reproduce.

m-mcgowan, 2024-08-23:
I see this routinely. Yesterday I had dozens of python processes all pinned (at 50%..probably because there were so many?), to the point that my Mac claimed it was out of memory.

When I opened my laptop this morning, I found 4 python processes running at %100 cpu. 

I used `sudo killall python`, at 9:56, which didn't do anything, so then tried `sudo killall -9 python` 3 times in succession. 

Back in VS Code, I see a notification "The Black Formatter server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information."

Logs are below. 

This could be completely unrelated to this github issue, and in hindsight I should have at least looked at all the python processes in the system in more detail, rather than using a blunt tool to kill them all. I will report in more detail when this happens again.

```
4-08-23 00:08:28.431 [info] Server run command: /Users/mat/.pyenv/versions/3.8.16/bin/python /Users/mat/.vscode/extensions/ms-python.black-formatter-2024.2.0/bundled/tool/lsp_server.py
2024-08-23 00:08:28.431 [info] Server: Start requested.
2024-08-23 00:08:30.991 [info] CWD Server: /Users/mat/e/firmware/iot_thing
2024-08-23 00:08:31.130 [info] Settings received on server:
[
    {
        "cwd": "/Users/mat/e/firmware/iot_thing",
        "workspace": "file:///Users/mat/e/firmware/iot_thing",
        "args": [],
        "path": [],
        "interpreter": [
            "/Users/mat/.pyenv/versions/3.8.16/bin/python"
        ],
        "importStrategy": "useBundled",
        "showNotifications": "off"
    }
]

2024-08-23 00:08:31.314 [info] Global settings received on server:
{
    "cwd": "/",
    "workspace": "/",
    "args": [],
    "path": [],
    "interpreter": [],
    "importStrategy": "useBundled",
    "showNotifications": "off"
}

2024-08-23 00:08:31.346 [info] sys.path used to run Server:
   /Users/mat/.vscode/extensions/ms-python.black-formatter-2024.2.0/bundled/libs
   /Users/mat/.vscode/extensions/ms-python.black-formatter-2024.2.0/bundled/tool
   /Users/mat/.pyenv/versions/3.8.16/lib/python38.zip
   /Users/mat/.pyenv/versions/3.8.16/lib/python3.8
   /Users/mat/.pyenv/versions/3.8.16/lib/python3.8/lib-dynload
   /Users/mat/.pyenv/versions/3.8.16/lib/python3.8/site-packages
2024-08-23 00:08:31.356 [info] /Users/mat/.pyenv/versions/3.8.16/bin/python -m black --version
2024-08-23 00:08:31.358 [info] CWD formatter: /Users/mat/e/firmware/iot_thing
2024-08-23 00:08:31.359 [info] Version info for formatter running for /Users/mat/e/firmware/iot_thing:
black, 24.3.0 (compiled: no)
Python (CPython) 3.8.16

2024-08-23 00:08:31.381 [info] SUPPORTED black>=22.3.0
FOUND black==24.3.0

2024-08-23 09:56:29.682 [info] [Error - 9:56:29 AM] Server process exited with signal SIGTERM.
2024-08-23 09:56:29.693 [info] [Info  - 9:56:29 AM] Connection to server got closed. Server will restart.
2024-08-23 09:56:29.693 [info] true
2024-08-23 09:56:30.470 [info] CWD Server: /Users/mat/e/firmware/iot_thing
2024-08-23 09:56:30.470 [info] Settings received on server:
[
    {
        "cwd": "/Users/mat/e/firmware/iot_thing",
        "workspace": "file:///Users/mat/e/firmware/iot_thing",
        "args": [],
        "path": [],
        "interpreter": [
[... 150 lines cut ...]
    "path": [],
    "interpreter": [],
    "importStrategy": "useBundled",
    "showNotifications": "off"
}

2024-08-23 09:56:37.289 [info] sys.path used to run Server:
   /Users/mat/.vscode/extensions/ms-python.black-formatter-2024.2.0/bundled/libs
   /Users/mat/.vscode/extensions/ms-python.black-formatter-2024.2.0/bundled/tool
   /Users/mat/.pyenv/versions/3.8.16/lib/python38.zip
   /Users/mat/.pyenv/versions/3.8.16/lib/python3.8
   /Users/mat/.pyenv/versions/3.8.16/lib/python3.8/lib-dynload
   /Users/mat/.pyenv/versions/3.8.16/lib/python3.8/site-packages
2024-08-23 09:56:37.293 [info] /Users/mat/.pyenv/versions/3.8.16/bin/python -m black --version
2024-08-23 09:56:37.293 [info] CWD formatter: /Users/mat/e/firmware/iot_thing
2024-08-23 09:56:37.416 [info] Version info for formatter running for /Users/mat/e/firmware/iot_thing:
black, 24.3.0 (compiled: no)
Python (CPython) 3.8.16

2024-08-23 09:56:37.416 [info] SUPPORTED black>=22.3.0
FOUND black==24.3.0

2024-08-23 09:56:37.635 [info] [Error - 9:56:37 AM] Server process exited with signal SIGKILL.
2024-08-23 09:56:37.637 [info] [Error - 9:56:37 AM] The Black Formatter server crashed 5 times in the last 3 minutes. The server will not be restarted. See the output for more information.
```

S0urceror, 2026-01-01:
Issue is still here. Just had to kill two hanging Python processes from a debug session in VSCode. Started heating up my laptop and draining my battery. In Activity Monitor both where at 100%. Killed them and all is fine now.