platformio/platformio-vscode-ide #2895: PIO update - cannot find working Python 3.6+ ....

AlesSt, 2022-01-21:
Description of problem
Starting VScode PIO goes into update and drops a Notification saying I have no Python installed.

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN

https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue

Been reading Python problems for last 3 days and none of the solutions help.......

Configuration

VSCode: 1.63.2
PIO IDE: v2.4.0
System: Win 10 pro x64 21H1 version

So I tried installing diff versions of Python to no success (ALWAYS added to PATH and restarted computer)
Last Python was by suggestion exe not msi distribution 3.8.8 (also in PATH) accessible in cmd everywhere.
In settings enabled from start to use internal python.
Also deleted python3 and penv folders to no help (python3 was recreated back but penv not)
In dev tools console I see in red ECONNREFUSED but ONLY to 127.0.0.1:80xx (xx = about 10 to 20 refuses to diff ports) => I guess those refuses are to pio home because it is not running.

Please help as after 3 days I am loosing all hope .....

AlesSt, 2022-01-21:
console output:

Error: PlatformIO Core is not installed	at U (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:6650)	at A.value (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:790047)	at A.value (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:789695)	at listOnTimeout (internal/timers.js:554:17)	at processTimers (internal/timers.js:497:7)
I @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:472
_invokeHandler @ rpcProtocol.ts:457
_receiveRequest @ rpcProtocol.ts:373
_receiveOneMessage @ rpcProtocol.ts:295
(anonymous) @ rpcProtocol.ts:160
fire @ event.ts:577
fire @ ipc.net.ts:638
_receiveMessage @ ipc.net.ts:958
(anonymous) @ ipc.net.ts:831
fire @ event.ts:577
acceptChunk @ ipc.net.ts:382
(anonymous) @ ipc.net.ts:338
P @ ipc.net.ts:60
emit @ events.js:315
addChunk @ internal/streams/readable.js:309
readableAddChunk @ internal/streams/readable.js:284
Readable.push @ internal/streams/readable.js:223
onStreamRead @ internal/stream_base_commons.js:188

console.ts:137 [Extension Host] Error: Could not start PIO Home server: Error: timeout	at c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9933
I @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:472
_invokeHandler @ rpcProtocol.ts:457
_receiveRequest @ rpcProtocol.ts:373
_receiveOneMessage @ rpcProtocol.ts:295
(anonymous) @ rpcProtocol.ts:160
fire @ event.ts:577
fire @ ipc.net.ts:638
_receiveMessage @ ipc.net.ts:958
(anonymous) @ ipc.net.ts:831
fire @ event.ts:577
acceptChunk @ ipc.net.ts:382
(anonymous) @ ipc.net.ts:338
P @ ipc.net.ts:60
emit @ events.js:315
addChunk @ internal/streams/readable.js:309
readableAddChunk @ internal/streams/readable.js:284
Readable.push @ internal/streams/readable.js:223
onStreamRead @ internal/stream_base_commons.js:188

console.ts:137 [Extension Host] rejected promise not handled within 1 second: Error: PlatformIO Core is not installed
I @ console.ts:137
$logExtensionHostMessage @ mainThreadConsole.ts:39
_doInvokeHandler @ rpcProtocol.ts:472
_invokeHandler @ rpcProtocol.ts:457
_receiveRequest @ rpcProtocol.ts:373
_receiveOneMessage @ rpcProtocol.ts:295
(anonymous) @ rpcProtocol.ts:160
fire @ event.ts:577
fire @ ipc.net.ts:638
_receiveMessage @ ipc.net.ts:958
(anonymous) @ ipc.net.ts:831
fire @ event.ts:577
acceptChunk @ ipc.net.ts:382
(anonymous) @ ipc.net.ts:338
P @ ipc.net.ts:60
emit @ events.js:315
addChunk @ internal/streams/readable.js:309
readableAddChunk @ internal/streams/readable.js:284
Readable.push @ internal/streams/readable.js:223
onStreamRead @ internal/stream_base_commons.js:188

console.ts:137 [Extension Host] stack trace: Error: PlatformIO Core is not installed	at U (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:6650)	at F (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:6842)	at c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9726	at new Promise (<anonymous>)	at aA (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9705)	at processTicksAndRejections (internal/process/task_queues.js:93:5)	at async Module.fA (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9449)	at async A.getWebviewContent (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\dist\extension.js:1:17735)	at async A.newPanel (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\dist\extension.js:1:17203)	at async A.toggle (c:\Users\aless\.vscode\extensions\platformio.platformio-ide-2.4.0\dist\extension.js:1:16768)	at async r._executeContributedCommand (c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:95:31072)

AlesSt, 2022-01-25:
as discussed in #2881 I uninstalled PlatformIO extension from VScode and deleted .platformio folder.
Restarted VScode and reinstalled PlatformIO. => no success
python3 folder is created but no penv.
I also tried per discussion again in #2881 to install python 3.9.8 (.exe file, not the embedded version => before I had 3.8.8) again to no success.

This thing is driving me crazy for already a week.... So PLEASE I really need some help to resolve this issue ....

Fa-b, 2022-01-25:
I find it strange that no penv folder is created on reinstall. Has it worked before on this device? Can it be that you customized the extension settings and they persist after deinstallation of PlatformIO?

AlesSt, 2022-01-25:
When i uninstalled ext i was already without penv as I saw somewhere to try and delete penv and python3 folders but anything I try I just get notiff no python 3.6+ ...... i did try the manual core cli install and it finishes without any err.

It is so confusing .... I am totally in darkness here ......

customizing you mean the PIO ext? in the JSON file or GUI settings?
I did some customizations but don't think any significant to install/update problems. 
so in JSON:
/* ----------------------------------- PIO ---------------------------------- */
"platformio-ide.forceUploadAndMonitor": true,
"platformio-ide.activateOnlyOnPlatformIOProject": true,
that's all

on the GUI part:
tick on => builtin PIO Core, Builtin Python
the rest is of no importance regarding the update/install (oh and no tick on Development Core also no custom PATH .. terminal in vsc recognizes python 3.9.8 nor Custom Py Pi Index Url )

and THANK you so much for reaching out :)  I ve picked up a few of your suggestions along the way about how to tackle this one but to no success :)

Fa-b, 2022-01-26:
Have you tried downgrading the PIO extension in VSCode to see if you can get an earlier version running?

AlesSt, 2022-01-26:
Hi, yes I managed to get up the earlier extension 2.3.1 (Core 5.2.4 / Home 3.4.0)

Now since I already messed everything up ----- I will try to upgrade PIO ext one step a time so I can check where everything stops perhaps that should give some more clues what would be the problem.

AlesSt, 2022-01-27:
So what I did (for people with the same problem)
In browser go to vscode marketplace => download version 2.3.4 and then vscode start 
code --force --install-extension platformio.platformio-ide-2.3.4.vsix     => no --force if you are not downgrading from newer version

for me now the version 2.3.4 works as supposed but NO WAY to install version 2.4.0 

also go to vscode settings and DISABLE extensions auto update (I left the update check ticked) otherwise platformio will go in auto update !!!!! please check every now and then and update other extensions :) 

If any one has any idea what could be the problem in my case (the python checkup in PIO installation looks like has a bug or something)  I STILL NEED HELP please :)

Fa-b, 2022-01-27:
You should also be able to just right-click `PlatformIO IDE` in the Installed Extensions list view in VSCode and select `Install Another Version`.
Glad you can at least work with an older version now @AlesSt , still strange that 2.4.0 causes these errors on your machine.

AlesSt, 2022-01-28:
Thank you very much for your effort and help :) I see now the right click option (smileyShootingHimselfInTheHead) :)

As usually I went the "hard" way :)

I guess something with the python checkup script isn't as is should be .... as I see a lot of ppl struggling with this transition to 2.4.0 and each one had almost different solution to solve it :) Mine is yet to be found I guess :D

closed 2022-02-11 by ivankravets