platformio/platformio-vscode-ide #2898: Start PIO Home Server

sexyhexy, 2022-01-23:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.59.1
  PIO IDE: v2.3.3
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

    at cmd (c:\Users\hexad\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:191:10)
    at ChildProcess.a (c:\Users\hexad\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\hexad\.vscode\extensions\platformio.platformio-ide-2.3.3\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

Fa-b, 2022-01-24:
see #2881

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

closed 2022-02-11 by ivankravets