platformio/platformio-vscode-ide #2902: Installation problem on VSCode

jerome6994, 2022-01-26:
Hello

I'm new to PlatformIO and just tried installing the one in VSCode.
Following the installation, I don't have a shortcut on the left and I don't have the Home bar at the bottom either!
I have tried several times to do this but nothing works.

I am on Windows 10 latest version
VSCode Latest version too
Python 2.7.17

I must be forgetting a few things but I don't know what!

Thank you for your help

Jerome

jerome6994, 2022-01-29:
[image]

I don't have the icon and I can't launch the application!

mike-berriman, 2022-02-05:
I was having the same problem.  What it turned out to be was an old version of VSCode.  Looking at the icons you have on the left, you might have the same problem.  Help->Check for Updates in VSCode was telling me that it was up to date, when in fact it was on version 1.27, and the current version is 1.64.  Once I upgraded, I re-installed PlatformIO and on first glance it now seems to be working perfectly.

ivankravets, 2022-02-05:
@jerome6994 have you resovled this issue?

jerome6994, 2022-02-05:
Bonjour

so I have the latest version in force it seems to me.
[image]

then to answer @ivankravets yes I just solved the problem in the following way.
I deleted all the elements present in .vscode and in .platformio in the user directory of windows.
I then uninstalled VSCode
I installed VSCode then installed the PlatformIO extension
I then created a new file that I named platformio.ini that I saved in the desired working directory
I quit VSCode then opened it again
I clicked on the created platformio.ini file
and the miracle is the Platform IO product is Installed!!!
It's still a curious process to have Platform IO

ivankravets, 2022-02-05:
Hm, strange issue. Glad to hear that you solved it.

Happy coding with PlatformIO!

closed 2022-02-05 by ivankravets

jerome6994, 2022-02-05:
Yes strange but I think I'm not the only one because I found a video on Youtube that talks about it

Definitely a bug to fix but I don't know where

ivankravets, 2022-02-05:
It would be great if someone give us a remote session to debug this issue. We can't reproduce it.