platformio/platformio-vscode-ide #2906: IDE not loading in VSCode

nicr4wks, 2022-01-27:
Hi all,

I had originally logged this at the platformio forums but was recommended to log here instead.
https://community.platformio.org/t/ide-not-loading-in-vscode/

Initial issue was I can install the the platformio extension but it never loads within vscode, I don't get the alien head on the left or bottom toolbars, I do not get the .platformio folder created.

I performed a clean install of vsc by uninstalling then manually deleting the following folders:

%LocalAppData%\Programs\Microsoft VS Code
%AppData%\Code
%UserProfile%\.vscode

I then re-downloaded vsc and installed, opened vsc and installed the platformio extension and it worked, the icon was available and the .platformio folder was created and populated. Upon closing and re-opening vsc the platformio extension will no longer load and icons are not available.

I can manually execute 'pio home' outside of vsc and it loads in my normal browser fine, I just can't get the extension to load within vsc.

Appreciate any ideas to get this going, thanks!

[image]

windy54, 2022-01-31:
I have had the same problem and was asked to re-post here rather than on the platform io community forum, so I have copied my original post below.

Hi,

I had been using VSC with platform IO on a raspberry PI 4 running raspberry pi os with no problems.

It was being used for ESP8266 development of the esp-rfid project.

I recently came back to it, I have not updated my OS from Buster and the platform io icon is not displayed in the sidebar and there are no options on the bottom taskbar for compile re-build etc.
When I click on extensions platform io is enabled globally.
No problems are reported in the terminal.
If I Uninstall platform io amd then install it, everything works, the icon is displayed in left hand side bar and I have the options on the bottom taskbar.
If I close and re-open VSC the icons are not displayed.
I have googfed the problem and tried right clicking on the taskbar,
The option “activate only on platform IOProject is not selected.
Other settings such as"Disbale PIOHome Startup” are not selected.
There are no updates available.
The same problem occurs with the latest Bullseye release of raspberry PI OS, which is why I reverted to what was my Buster working installation.
I have also re-installed VSC and platform IO and cannot fix the problem.

Has anyone a suggestion as to what to try next.

Hopefully I have posted this in the correct place.

Steve

ivankravets, 2022-02-05:
It is a very strange issue. Have you resolved it?

windy54, 2022-02-05:
Not yet

Cheers
Steve Gale

ivankravets, 2022-02-05:
Could you contact me directly via me@ikravets.com and help to debug this issue?

AI4sens, 2022-04-10:
I have exactly the same problem> I deleted also the app and cleaned the registry but the problem stays.

closed 2022-06-18 by stale[bot]