platformio/platformio-vscode-ide #2907: Start PIO Home Server

Fr0sTn, 2022-01-27:
%23 Description of problem
 The Home page not loading up and an error message as below popping up

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.3.4
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\nazar\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:5850
	at ChildProcess.a (c:\Users\nazar\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:771396)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\nazar\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

Fr0sTn, 2022-02-05:
Hello!
I have not been able to resolve the issue yet.
I've had PlatformIO for more than a year now.
When I just had it installed, everything worked fine, but I haven't used it in some time after.
When I tried using it again for my next project about two weeks ago, it started giving me the error posted above. 
My bet is: I've probably misplaced some files by accident some time ago or made some other modifications to my system that conflict with PlatformIO, but I have no idea what it could be, as I can't even track it down at this point.

Fa-b, 2022-02-06:
Since you are unsure if you have made changes that led to the issue, can you follow the steps I listed [here](https://github.com/platformio/platformio-vscode-ide/issues/2881#issuecomment-1019914493) and see if that solves it for you too? Make sure to keep a copy of the original `.platformio` folder to see if the error can be reproduced by swapping them later again.

Fr0sTn, 2022-02-06:
Problem fixed, thanks for your help. No idea, how I haven't come across this link before reporting the error, as it now seems to me that it's quite a common issue.

closed 2022-02-06 by Fr0sTn

Fa-b, 2022-02-06:
Glad this helped. Can you reproduce the error by changing the folder back to the faulty version? No worries you would not need to reinstall anything if you keep a copy of the working folder and repöace them again afterwards, only restart VSCode after making changes in the directory. If the error is reproducible, could you compare the contents of `.platformio/penv/pyvenv.conf` To help troubleshooting the issue?

Fr0sTn, 2022-02-08:
I just reproduced the error by restoring the faulty .platformio folder, as requested, and compared the two pyvenv.conf files. Please see the contents from the two conf files below and don't hesitate to let me know if I can be of more help.

Faulty pyvenv.conf from the old folder:

home = C:\Users\nazar\.platformio\python3
include-system-site-packages = false
version = 3.7.7

Working pyvenv.conf from the recently generated folder:

home = C:\Users\nazar\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0
include-system-site-packages = false
version = 3.10.2

Fr0sTn, 2022-02-08:
> could you compare the contents of `.platformio/penv/pyvenv.conf` To help troubleshooting the issue?

-----

I just reproduced the error by restoring the faulty .platformio folder, as requested, and compared the two pyvenv.conf files. Please see the contents from the two conf files below and don't hesitate to let me know if I can be of more help.

Faulty pyvenv.conf from the old folder:

home = C:\Users\nazar.platformio\python3
include-system-site-packages = false
version = 3.7.7

Working pyvenv.conf from the recently generated folder:

home = C:\Users\nazar\AppData\Local\Microsoft\WindowsApps\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0
include-system-site-packages = false
version = 3.10.2

ivankravets, 2022-02-08:
Did you install Python 3.10.2 manually?

ivankravets, 2022-02-08:
Do you have a backup of a broken `.platformio` folder? Could you reproduce this issue again?

Fa-b, 2022-02-08:
@ivankravets that‘s comfirmed already see above. To reproduce the error one does not even have to replace the entire .platformio folder, keeping the local python folder within it and manually linking to that in pyvenv.conf is sufficient.

ivankravets, 2022-02-08:
@Fa-b I tried today different cases in VM and could not reproduce the issue. Could you help me to debug this issue via a remote session? me@ikravets.com

Fa-b, 2022-02-08:
I don‘t have access to the machine with this error right now. My device at home with also python 3.7.7 local version does not produce this error (yes python -V in VSCode also yields 3.7.7). I will both carry the working folder over to the „faulty machine“ see if that works and carry the „faulty folder“ to the working machine see if that not works and get back to you tomorrow if thats alright.

Fr0sTn, 2022-02-09:
Perhaps I could upload the old .platformio folder for you two to have a look at? I still have it in my trash bin. 
Please don't hesitate to poke me in the back if anything else is needed.

ivankravets, 2022-02-10:
Could you copy  `.platformio` folder from trash to any place and 
```
cd .platformio/penv/Scripts
pio home
```

Does it work?

Fa-b, 2022-02-11:
indeed it works, but only as long as `pyvenv.cfg` points to python 3.10.2 systemwide installation. if I change to the faulty 3.7.7 version it will fail no matter from where I execute it. Any command (also pio -h) will lead to that exact same error. Bringing a 'working' python 3.7.7 instance (from other machines install) does not resolve the issue though, so I guess rather than a corrupt installation something in the pio executables might expect a higher python version because internally it thinks it provided something? I wasn't yet able to test the faulty files on the working machine.