platformio/platformio-vscode-ide #2909: Start PIO Home Server

sexyhexy, 2022-01-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

    at cmd (c:\Users\hexad\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:195:11)
    at ChildProcess.r (c:\Users\hexad\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\hexad\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

sexyhexy, 2022-02-05:
Hi Ivan,
Thanks for the reply.
Yes, I uninstalled pio and cleared out any references then re-installed and all was well. 
Working great again and FYI I have just started using the debugger with an ESP-Prog and it works brilliantly.
Thanks for all your efforts.
Roy

closed 2022-02-05 by sexyhexy