platformio/platformio-vscode-ide #2915: Start PIO Home Server

GNAKACH, 2022-01-28:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\Guy-Franck\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\Guy-Franck\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\Guy-Franck\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

GNAKACH, 2022-02-06:
Hi
Thanks for your message.
I don't think the issue is resolved but after rebboting and starting afresh
i am able to run my project.
It seems the issue arose when i changed platforms  etween 2 projects
(arduino and espressif).
Thanks and regards

Guy-Franck

ivankravets, 2022-02-07:
Could you reproduce it now?

GNAKACH, 2022-02-07:
Will try when i get back home in a couple of days.

closed 2022-02-11 by ivankravets