platformio/platformio-vscode-ide #2922: Start PIO Home Server

labmaster, 2022-01-31:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\meyer\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\meyer\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\meyer\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

labmaster, 2022-02-05:
The only way to solve it (and i tried different things a whole day long)  was to fully uninstall PlatformIO including removing all files in the directory structure, then fully uninstall VSC even manually removing installation files and after that, starting all over again from the scratch.
And yes i already used PlatformIO since quite a while.

closed 2022-02-11 by ivankravets