platformio/platformio-vscode-ide #2923: Start PIO Home Server

mottibz, 2022-02-01:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\Motti\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\Motti\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\Motti\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

mottibz, 2022-02-05:
Hi Ivan, I am an IoT Maker and a PlatformIO user for a long time and had no problem until recently.
The error started to consistently appear when starting VSC. I tried to remove both PlatformIO from VSC and reinstall and also reinstalled VSC, with no luck.
The only thing that helped was after I deleted the .platformio directory (actually, renamed to something else) and this caused the creation of a new clean one by PlatformIO.
Looks like the directory was corrupted in some way and now after re-creating the directory, all works fine...
If you want, I can zip the old directory and upload for you to investigate.
Thanks for a great development environment!!!

ivankravets, 2022-02-05:
Could revert back to the old folder and reproduce this issue?

mottibz, 2022-02-05:
I am not at home right now. Will give it a try hopefully tomorrow and let you know.

Fa-b, 2022-02-06:
Could you also check for differences in the `.platformio/penv/pyvenv.conf ` files while you‘re at it?

numanair, 2022-02-11:
> Could you also check for differences in the .platformio/penv/pyvenv.conf files while you‘re at it?

For me the only difference was "version = 3.7.7" for the non-working folder and "version = 3.9.10" for the newly created folder. Changing this back did not reproduce the problem. Moving the old folder back _did_ reproduce the problem.

ivankravets, 2022-02-11:
@dalander do you still have "old" folder? Could you revert back to it and reproduce this issue?

> link points into 404,

Where did you find this link?

dalander, 2022-02-11:
Link: The error pops up with a with a message that leads into creating a github issue. Think this is a feature from VS Code. Was near to just create a new issue, too.

Unfortunately in this moment at testing I destroyed the backup folder of the platformio installation. BUT: I played around with changing folders from old to new installation vice versa, and was able to reproduce and, lets say, deproduce / solve the issue by just changing some files (without touching directories):
Content of platformio\penv\Lib\site-packages contained match files in the fault installation. I removed all files, and copied the only files of the new installation:

bottle.py
distutils-precedence.pth
tabulate.py
typing_extensions.py

And it worked. Copied back all files from faulty installation, the issue comes back. 
Really sad to say that a later restart then removed all backup folders here, and I am not able to recreate this content. But can say that from a quick look also there were more folders in than in the new installation.

dalander, 2022-02-11:
Not sure if 100% correct, at one point I had the impression that the python versions between 
.platformio/python3
and
.platfromio/pyenv/Scripts 
were different (3.7.7 vs 3.9.10). Probably only a changed Timestamp initiated a reinstall, now both versions are the same (3.9.10)

ivankravets, 2022-02-11:
Yes, there is a newer version of portable Python in the registry, this is the reason why you see 3.9.10.

Sorry, we can't solve this issue because can't reproduce it. I don't use Windows but tried it in different VMs with Win 10. Everything works great.

If someone can provide a remote session to debug this issue, we would be thankful.

dalander, 2022-02-11:
Yes, understand. For me the issue is solved. 
May anotherone with the same issue can be asked for the content of the **.platformio\penv\Lib\site-packages** folder . This was defenitly the part wich causes the error in my installation.
Thanks for your support, and the great work!

ivankravets, 2022-02-11:
Did you also install PlatformIO so long and didn't use it?

dalander, 2022-02-11:
Yes, probably there were missed updates in between. Last time I used it and it worked was on 27.11.2021

numanair, 2022-02-11:
@ivankravets Do you want my .platformio folder that has the issue?

ivankravets, 2022-02-11:
@numanair could you reproduce this issue on your machine?

numanair, 2022-02-11:
Yes, I can reproduce the issue by replacing the .platformio folder with my backup of the "bad" one.

ivankravets, 2022-02-11:
@numanair please email me directly via me@ikravets.com. Thanks!

closed 2022-02-11 by ivankravets

ivankravets, 2022-02-11:
@numanair thank you so much for providing a session where we could reproduce this issue. The issue is linked with a broken https://github.com/encode/starlette package on which PlatformIO Core depends.

The hotfix from our side is provided and a new version of PlatformIO IDE for VSCode is available. Please update.

MichaelGroesel, 2022-02-14:
%23 Description of problem

 Dear Ivan,

thank you for your message. I updated VSCode and PIO IDE to the followed version. On my german win 10 professional notebook PIO Home works, but on my english win 10 enterprise PC the PIO Home does not work, it displayed every time "Loading..." and followed error occur. Only the last Version of PIO IDE V2.3.1 works. Please inform me about a solution of this problem. Thank you in advance.

Best regards

Michael

  %23 Configuration

  VSCode: 1.64.2
  PIO IDE: v2.4.2
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Could not start PIO Home server: Error: getaddrinfo ENOTFOUND http://127.0.0.1
	at c:\Users\mg22\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\platformio-node-helpers\dist\index.js:1:9933
	at runMicrotasks (<anonymous>)
	at processTicksAndRejections (internal/process/task_queues.js:93:5)
  ```

commit 29a302f1 (2022-02-11): Fixed an "Error: No such command "home" when opening PIO Home // Resolve #2923