platformio/platformio-vscode-ide #2925: Debugging does not start

maxgerhardt, 2022-02-04:
In the last 3 days, there has been an unusual concentration of topics about users not being able to debug. The symtomps are:
* Debug Console empty after starting debugging
* debugging works in the CLI when invoked via `pio debug --interface=gdb -x .pioinit`
* all reports use Windows, some say on Linux it works

These topics are:
* https://community.platformio.org/t/debugging-arduino-platformio-and-windows/25777
* https://community.platformio.org/t/debugging-not-starts/25797
* https://community.platformio.org/t/debugger-stopped-working-in-vscode/25802

I myself am running VSCode 1.63.2 and do not have problems, some failure report say they're running VSCode 1.64.0.

There may be a VSCode update that breaks debugging? In any case, a concentration of these reports over a small period of time is too unusual.

CC @ivankravets.

maxgerhardt, 2022-02-04:
Hm I've updated my VSCode version to 1.64.0 and I could still debug. Using the release version (1.8.2) the Microsoft C/C++ extension also works, just like the current pre-release version (1.8.3). There has to be something else in play here.

FaBjE, 2022-02-04:
I (might be) experiencing the same issue. I have been investigating it for the past couple of days. It's driving my crazy.
I used to develop in a virtual machine. But I had USB issues with that so decided to do an install in my host os.
But I'm unable to get (the same project and setup) working in VSCode under my native OS.
If you need any more information I'm happy to help.

`pio debug --interface=gdb -x .pioinit` works fine in both cases.

**My setups:**
- VM Windows 10: Working  [Installed around 2022-01-10]
- Native Windows 10: Not working
- VM Linux: Working [Installed around 2022-01-6] (Not investigated further for this issue)

### Tests

**With "showDevDebugOutput": true, I get the following output**

_VM Win10 (Working)_ 
```
1-gdb-set target-async on
2-interpreter-exec console "source .pioinit"
3-enable-pretty-printing
undefinedOpen On-Chip Debugger  v0.11.0-esp32-20211220 (2021-12-20-15:43)
Licensed under GNU GPL v2
bla bla bla
```

_Native Win10 (Not working):_
```
1-gdb-set target-async on
2-interpreter-exec console "source .pioinit"
3-enable-pretty-printing
-gdb-exit
```

**Python:**
I noticed that my VM's were using python-package: 1.30908.0 and my new Native Win10 install 1.30910.0
I installed python 3.9.8 on my machine and re-installed platformIO using that version (it wasn't easy) but it made no difference.

_VM Win10 (Working)_ 
Portable (buildin)  1.30908.0 

_Native Win10 (Not working):_
Portable (buildin)  1.30908.0 

**VSCode**
My only extensions are PlatformIO + C/C++. No other extensions are installed.
VSCode updated itself this morning. It made no difference.

_VM Win10 (Working)_ 
```
Version: 1.64.0 (user setup)
Commit: 5554b12acf27056905806867f251c859323ff7e9
Date: 2022-02-03T04:22:20.678Z (1 day ago)
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.17763
```

_Native Win10 (Not working):_
```
Version: 1.64.0 (user setup)
Commit: 5554b12acf27056905806867f251c859323ff7e9
Date: 2022-02-03T04:22:20.678Z (1 day ago)
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19044
```

**PlatformIO**
See version info below. I switched to the dev version but it made no difference.

_VM Win10 (Working)_ 
```
pio system info
--------------------------  -------------------------------------------------------
PlatformIO Core             5.2.4
Python                      3.9.8-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\IEUser\.platformio
PlatformIO Core Executable  C:\Users\IEUser\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\IEUser\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       2
Tools & Toolchains          15
```

_Native Win10 (Not working):_
```
pio system info
--------------------------  ------------------------------------------------------
PlatformIO Core             5.2.4
Python                      3.9.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Fabje\.platformio
PlatformIO Core Executable  C:\Users\Fabje\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\Fabje\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          12
--------------------------  ------------------------------------------------------
```

**Debug server**
Figuring the problem must be somewhere around starting GDB. I set my 
`debug_server = C:/Windows/Notepad.exe` and noticed some funny behaviour

_Native Win10 (Not working):_
Notepad pops up but immediatly exitst/gets killed.

_VM Win10 (Working)_ 
Notepad opens and stays open.

maxgerhardt, 2022-02-04:
The non-working setup has Windows version `OS: Windows_NT x64 10.0.19044` and the working setup `OS: Windows_NT x64 10.0.17763`? I might try to install all pending Windows 10 updates and see if that breaks it.

ivankravets, 2022-02-04:
@maxgerhardt thanks for filing this issue. The strange issue which we can't reproduce. 

@FaBjE yes, we updated portable Python to 3.9.10 a few days ago https://registry.platformio.org/tools/platformio/python-portable/versions

Maybe this is the issue?

@maxgerhardt could you try to remove `%USER_HOME%/.platformio/python3` and `%USER_HOME%/.platformio/penv`? The IDE will install the latest 3.9.10. Does it work for you?

maxgerhardt, 2022-02-04:
I'm running PlatformIO from my system Python version, 3.8.6. I'll uninstall my global installation and reinstall with the portable python.

maxgerhardt, 2022-02-04:
**WOW**, it broke.

Debugging controls (play, step into, ..) briefly pops up then disappears immediately. "Terminal" shows a successfull debug build, the "Debug Console" is **empty**. This really breaks it.

```
PS C:\Users\Max\Desktop\gd32_dev\gd32-pio-projects\gd32-spl-blinky> pio system info
--------------------------  ----------------------------------------------------
PlatformIO Core             5.2.5a7
Python                      3.9.10-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp65001
PlatformIO Core Directory   C:\Users\Max\.platformio
PlatformIO Core Executable  C:\Users\Max\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\Max\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       59
Tools & Toolchains          204
--------------------------  ----------------------------------------------------
```

FaBjE, 2022-02-04:
> The non-working setup has Windows version `OS: Windows_NT x64 10.0.19044` and the working setup `OS: Windows_NT x64 10.0.17763`? I might try to install all pending Windows 10 updates and see if that breaks it. My machine is on `Microsoft Windows [Version 10.0.19044.1466]`.

I tried to, but its a VM image from here: https://developer.microsoft.com/en-us/microsoft-edge/tools/vms/
It has no updates available.

> @FaBjE yes, we updated portable Python to 3.9.10 a few days ago https://registry.platformio.org/tools/platformio/python-portable/versions
> 
> Maybe this is the issue?

Well, doing a manual install on my native windows using python-3.9.8-amd64.exe It made no difference. (maybe I did something wrong)

**Gotcha!**
I deleted the python3 and penv folder in my VM now and let it re-install. now it is broken.
Exactly same behaviour as my native machine.

Is there a (easy) way to force platformio to install using the older python portable package?

maxgerhardt, 2022-02-04:
I'm currently downloading the 3.7.7 version from https://dl.registry.ns1.platformio.org/tools/b4/89/747ab34b90655132013e722813783e77a8632c5d4d468fe92209a3e953e2/python-portable-windows_amd64-1.30707.0.tar.gz, deleting my `C:\Users\<user>\.platformio\python3` folder and extracting the old python version from above in that folder..

FaBjE, 2022-02-04:
You need to delete the virtual environment as well. 
Just replacing it in the python3 folder didn't work for me.
But the PIO installer from vscode is very stubborn in using the portable python from the repository.
Even when setting the "don't use it" setting and filling the "where is python" prompt it still used the portable python it downloaded itself in my setup.

maxgerhardt, 2022-02-04:
Okay, doing that what I wrote above gets me Python 3.7.7, which is also visible in `pio system info`, but debugging still immediately aborts like before.

````
> pio system info
--------------------------  ----------------------------------------------------
PlatformIO Core             5.2.5a7
Python                      3.7.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp65001
PlatformIO Core Directory   C:\Users\Max\.platformio
PlatformIO Core Executable  C:\Users\Max\.platformio\penv\Scripts\platformio.exe 
Python Executable           C:\Users\Max\.platformio\penv\Scripts\python.exe     
Global Libraries            0
Development Platforms       59
Tools & Toolchains          204
--------------------------  ----------------------------------------------------
```

After also removing my `C:\Users\<user>\.platformio\penv` folder and restarting VSCode, the PlatformIO core installs itself again and asks for a refresh. Doing that and attempting to debug again, it still does **not** work. I'll try downgrading the core version...

maxgerhardt, 2022-02-04:
Hm. I've tricked PlatformIO into using the 3.7.7 interpreter by keeping the `package.json` for the 3.9.10 version in the `python3` folder but replacing all files from the 3.9.10 download. Still did not work. In the penv, I uninstalled my previous 5.2.5a7 core version and reverted back to 5.2.3, the second-last stable. Still *does not work*. I can't seem to find what makes it work again.. Which is weird since all it re-installed was Python and the core version. I'll revert back to my global PlatformIO installation and not use the PlatformIO portable python.

ivankravets, 2022-02-04:
The broken Python 3.9.10 was removed from the registry. See the updated solution in the original comment https://github.com/platformio/platformio-vscode-ide/issues/2925#issue-1123992952

P.S: We will investigate what is going wrong with Python 3.9.10.

maxgerhardt, 2022-02-04:
One thing I can say is that when I reinstall my global PlatformIO installation and don't use the built-in interprter (PlatformIO extension settings), I'm back to being able to debug again.

[image]

I'll use the builtin Python again now that the 3.9.10 package was removed and I'll report back.

maxgerhardt, 2022-02-04:
Hm no, this is still not working for me. After removing `%USER_HOME%/.platformio/python3` and `%USER_HOME%/.platformio/penv` and `%USER_HOME%/.platformio/.cache` and restarting VSCode again, PlatformIO reinstalls itself with the Python 3.9.8 version and the latest PlatformIO core version, but debugging still aborts just as it previously did.

```
> pio system info
--------------------------  ----------------------------------------------------
PlatformIO Core             5.2.5a7
Python                      3.9.8-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp65001
PlatformIO Core Directory   C:\Users\Max\.platformio
PlatformIO Core Executable  C:\Users\Max\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\Max\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       59
Tools & Toolchains          204
--------------------------  ----------------------------------------------------
```

@FaBjE  does it work for you?

maxgerhardt, 2022-02-04:
Downgrading the PlatformIO core back to the stable 5.2.4 also makes no difference. I have no idea what's going on there.

FaBjE, 2022-02-04:
> @FaBjE does it work for you?

I Deleted the .cache, penv and python3 from .platformio in the user directory.
Let PIO re-install itself from visual studio But my VM setup is still broken.

Either this was not the cause or did some "permanent damage" to something else.

Can we re-open this issue?

ivankravets, 2022-02-04:
We are working on solving this issue...

geogeim, 2022-02-04:
i'm experiencing the same issue. deleting those 3 folders and waiting for reinstall didn't solve it for me
tried with both boards.

```
PS C:\Users\ike\Documents\PlatformIO\Projects\220204-005444-arduino-blink> pio system info
--------------------------  ----------------------------------------------------
PlatformIO Core             5.2.4
Python                      3.9.8-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\ike\.platformio
PlatformIO Core Executable  C:\Users\ike\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\ike\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          8
--------------------------  ----------------------------------------------------
```

```
[env]
platform = ststm32
framework = arduino
build_type = debug
debug_tool = stlink
upload_protocol = stlink

[env:bluepill_f103c8]
board = bluepill_f103c8
board_build.mcu = stm32f103c8t6
board_build.f_cpu = 72000000L

[env:blackpill_f401cc]
board = blackpill_f401cc
board_build.mcu = stm32f401ccu6
board_build.f_cpu = 84000000L
```

```
Version: 1.64.0
Commit: 5554b12acf27056905806867f251c859323ff7e9
Date: 2022-02-03T04:22:20.678Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19043
```

maxgerhardt, 2022-02-04:
I did a full removal of the `.platformio` folder and let it reinstall itself (with the PlatformIO Python interpreter), after reopening VSCode, the issue remains. I think only the PlatformIO devs can solve this at this point.

FaBjE, 2022-02-04:
I completely uninstalled VSCode + PlatformIO
After uninstalling vscode, removed
    %UserProfile%\.vscode
    %AppData%\Code
    %LocalAppData%\Programs\Microsoft VS Code

Removed
%LOCALAPPDATA%\pip

Installed everything from scratch

Exported python requirements file from working VM
https://pip.pypa.io/en/stable/cli/pip_freeze/#examples
Imported requirements file in Native Win10.
Still not working...

@maxgerhardt 
Maybe you should remove the "solution" part in your first post?

maxgerhardt, 2022-02-04:
Well Ivan edited that text in, not me :D I've reverted it back.

ivankravets, 2022-02-04:
The issue has been fixed, please update the extension to the latest version. It was not linked with Python, we reverted back the latest portable Python to 3.9.10.

The issue was linked with updated VSCode's Node.JS and "access violation" https://github.com/nodejs/node/blob/842e5a10ac59c826f875ef79e92d144e2655cc9a/test/common.js#L502-L505

FaBjE, 2022-02-04:
Thank you for explaining! I updated my extension and it is working now. Debugger starts as expected.

What is this "platformio-vscode-debug": "~1.4.1" dependency you changed?

ivankravets, 2022-02-04:
> What is this "platformio-vscode-debug": "~1.4.1" dependency you changed?

This is our front-end debugger for the VSCode. To fix this issue, we enabled `shell=true` for Node.JS ChildProcess.spawn on Windows.

FaBjE, 2022-02-04:
> > What is this "platformio-vscode-debug": "~1.4.1" dependency you changed?
> 
> This is our front-end debugger for the VSCode. To fix this issue, we enabled `shell=true` for Node.JS ChildProcess.spawn on Windows.

I see now. is it me or are the sources not on github?

If I can make a suggestion, It would be nice if just one output/printline would be added to the "Debug Console" something like "- Starting debug session -" (maybe with the time and date) so you can see it is actually (trying) to start.
It would be very helpfull for an enduser like me to determine where it cuts out. (All this time I had to figure from a disappearing progress bar and a empty "Debug Console")
If possible some more error information is always welcome ;)

Thank you for fixing this so quick! Have a nice weekend.

AnroM94, 2024-03-15:
Hi there, 

Seems like this is a problem again. My debugger is not starting.

BernardG, 2024-11-16:
still not working for me here, with PlatformIO core 6.1.16 and home 3.4.4. Impossible to start a debug session. Here is the error I get:

========================= [SUCCESS] Took 3.12 seconds =========================
Reading symbols from C:\Users\berna\Documents\PlatformIO\Projects\VariablesUNO\.pio\build\uno\firmware.elf...
done.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = avr-stub
PlatformIO: Initializing remote target...
.pioinit:13: Error in sourced command file:
:3333: Le syst�me a tent� de joindre un lecteur � un r�pertoire stock� sur un lecteur joint.

To translate the french message, it would be something like "system attempted to join a drive ? a folder stored on a linked drive".. What directory/drive could that be ? The board is Arduino Uno. Avrdude has no poblem.

maxgerhardt, 2024-11-16:
With `avr-stub`, it should not connect to port `:3333` at all: avr-gdb should connect to the serial port. What `platformio.ini` do you have? Does it follow https://docs.platformio.org/en/stable/plus/debug-tools/avr-stub.html#configuration?

BernardG, 2024-11-16:
it was not, as I did not know. Now it does, and I get a different error:

PlatformIO: debug_tool = avr-stub
PlatformIO: Initializing remote target...
Ignoring packet error, continuing...
warning: unrecognized item "timeout" in "qSupported" response
2
Ignoring packet error, continuing...
.pioinit:13: Error in sourced command file:
Bogus trace status reply from target: timeout

There is also a windows dialog saying:
"Failed to launch GDB: .pioinit: 13: error in sourced command file:
Bogus trace status reply from target: timeout (from interpreter-exec console "source .pioinit")

maxgerhardt, 2024-11-16:
Are you 100% sure you're running the required `src/main.cpp` code? This error says essentially that there is no GDB server answering, which is supposed to run on the Arduino. As from the documentation
```cpp
#include "Arduino.h"
#include "avr8-stub.h"
#include "app_api.h" // only needed with flash breakpoints

void setup()
{
  // initialize GDB stub
  debug_init();
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(1000);
  digitalWrite(LED_BUILTIN, LOW);
  delay(1000);
}
```

BernardG, 2024-11-16:
adding debug_init() creates an error. Where is the documentation about debugging with PlatformIO?  I have been looking for it, but was unable to find it.. My guess is that the library jdolinay/avr-debugger should be installed, not sure how. PlatformIO do so many things automatically, not sure what I have to do that it does not... :-)

maxgerhardt, 2024-11-16:
What error exactly? The documentation says to add
```
; GDB stub implementation
lib_deps =
    jdolinay/avr-debugger @ ~1.4
```
so that makes the avr8-stub.h and app_api.h headers available.

BernardG, 2024-11-16:
here is the error I have:

```
Linking .pio\build\uno\firmware.elf
HardwareSerial0.cpp.o (symbol from plugin): In function `Serial':
(.text+0x0): multiple definition of `__vector_18'
avr8-stub.c.o (symbol from plugin):(.text+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\uno\firmware.elf] Error 1
```
And yes, I modified  platformIO.ini with lib_deps as indicated.

BernardG, 2024-11-16:
Thanks for your help, I will look at all this more in depth. Really appreciate your time.

commit 7e3fe2fc (2022-02-04): Fixed an issue when debugging does not start on Windows // Resolve #2925