platformio/platformio-vscode-ide #2927: Start PIO Home Server

flyturbo, 2022-02-05:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.0
  PIO IDE: v2.4.1
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\bruno\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\bruno\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\bruno\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-05:
Have you resolved this issue? Have you had PlatformIO installed before?

flyturbo, 2022-02-05:
Hi,

yes, I have and it only happens when on load VS code. The PIO home page doesn't load anymore. It's just annoying more than anything else.

ivankravets, 2022-02-05:
Could you reproduce this issue again?

flyturbo, 2022-02-05:
Yes, I can...

[cid:540a77ce-fdeb-4301-b196-d9b9c7f7a42e]

ivankravets, 2022-02-05:
Could you contact me directly via me@ikravets.com? I would be thankful if you help us to debug this issue.

sexyhexy, 2022-02-05:
A complete re-install solved it for me!

ivankravets, 2022-02-05:
Ah, it really bad that you reinstalled everything. I wanted to reproduce this issue on your machine.

closed 2022-02-05 by ivankravets

sexyhexy, 2022-02-05:
Sorry about that Ivan!