platformio/platformio-vscode-ide #2930: Start PIO Home Server

lamloei, 2022-02-07:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.2
  PIO IDE: v2.4.0
  System: Windows_NT, 6.3.9600, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\admin\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Socket.<anonymous> (internal/child_process.js:439:11)
	at Socket.emit (events.js:315:20)
	at Socket.EventEmitter.emit (domain.js:467:12)
	at Pipe.<anonymous> (net.js:673:12)
  ```

ivankravets, 2022-02-07:
Have you resolved this issue? Could you help us to debug it?

zhang-program, 2022-02-11:
I have this issue too.
Please solve it!!!
Thanks.

ivankravets, 2022-02-11:
@zhang-program a temporary solution:

- Close VSCode
- Remove `c:\Users\xxx\.platformio\penv` folder
- Remove `c:\Users\admin\.platformio\python` folder
- Restart VSCode.

closed 2022-02-11 by ivankravets