platformio/platformio-vscode-ide #2932: Start PIO Home Server

SkylionFPV, 2022-02-08:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.0
  PIO IDE: v2.4.1
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\Misirycord\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\Misirycord\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\Misirycord\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-08:
Have you resolved this issue? Could you help us to debug it?

SkylionFPV, 2022-02-08:
I have not solved it yet and I am unsure of what is causing it I use
vscode/platformio to compile marlin 3d printer firmware mostly I was in the
middle of compiling firmware for my printer and it just stopped working I
don't understand why sorry

ivankravets, 2022-02-08:
Could you contact me directly via me@ikravets.com to help debug this issue?

closed 2022-02-11 by ivankravets