platformio/platformio-vscode-ide #2941: Start PIO Home Server

zhang-program, 2022-02-11:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.2
  PIO IDE: v2.4.1
  System: Windows_NT, 10.0.22533, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

    at cmd (c:\Users\loren\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:195:11)
    at ChildProcess.r (c:\Users\loren\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\loren\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

Please solve it!!!
Thanks.

Charder-uGym, 2022-02-11:
I just encounter the same problem.  I have projects with many platforms/boards/libraries combination. With @natedogg2020 method, will the deleting the whole .platformio folder cause side effects or I would need to re-install all the libraries again?

natedogg2020, 2022-02-11:
@Charder-uGym , you should be able to just move your .platformio folder into another directory, then reinstall platformio, then copy back the various libraries and files you originally had and need. I think the problem was with the portable python libraries.

Charder-uGym, 2022-02-11:
@natedogg2020 Thank you for your kind reply. It makes me feel more comfortable.

closed 2022-02-11 by ivankravets