platformio/platformio-vscode-ide #2973: Start PIO Home Server

mauriciokaster, 2022-02-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.1
  PIO IDE: v2.4.2
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\mauri\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.c (c:\Users\mauri\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\mauri\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-14:
Can you reproduce this issue if restart VSCode?

ivankravets, 2022-02-15:
c:\Users\lucal\.platformio\penv

closed 2022-02-15 by ivankravets