platformio/platformio-vscode-ide #2974: Start PIO Home Server

llandi, 2022-02-13:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.2
  PIO IDE: v2.4.2
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

    at cmd (c:\Users\lucal\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:195:11)
    at ChildProcess.c (c:\Users\lucal\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\lucal\.vscode\extensions\platformio.platformio-ide-2.4.2\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-14:
Could you reproduce this issue if restart VSCode?

llandi, 2022-02-14:
Yes,  but I noticed that it depends on the project I open. Also every now
and then I get the message that Python has crashed
Luca Landi (Genova-Italy)

ivankravets, 2022-02-15:
Please close VSCode, remove `c:\Users\lucal\.platformio\penv` and `c:\Users\lucal\.platformio\python3` folders, restart VSCode. Should work now.

closed 2022-02-15 by ivankravets