platformio/platformio-vscode-ide #2976: Start PIO Home Server

ColinHMurdoch, 2022-02-14:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.54.2
  PIO IDE: v2.3.4
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\Colin\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:5850
	at ChildProcess.a (c:\Users\Colin\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:771396)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:483:12)
	at ChildProcess.cp.emit (c:\Users\Colin\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
  ```

ivankravets, 2022-02-15:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/2923

Please update the VSCode and PlatformIO IDE extension for VSCode to the latest versions, and restart VSCode.

closed 2022-02-15 by ivankravets

ColinHMurdoch, 2022-02-15:
Thanks.

I have downloaded the VSCode update file but windows says the .zip file is invalid and will not open it.
What can I do to fix it?

Regards,

Colin Murdoch.