platformio/platformio-vscode-ide #2977: Start PIO Home Server

DALAA-MedAmine, 2022-02-14:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.1
  PIO IDE: v2.4.1
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\MohammedDalaa\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\MohammedDalaa\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\MohammedDalaa\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-02-15:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/2923

Please update the VSCode and PlatformIO IDE extension for VSCode to the latest versions, and restart VSCode.

closed 2022-02-15 by ivankravets

DALAA-MedAmine, 2022-02-15:
Yes i solved the problem, it was just an update ! thank you