platformio/platformio-vscode-ide #2985: Start PIO Home Server

ColinHMurdoch, 2022-02-15:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.54.2
  PIO IDE: v2.3.4
  System: Windows_NT, 10.0.19042, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\Colin\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:5850
	at ChildProcess.a (c:\Users\Colin\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\platformio-node-helpers\dist\index.js:1:771396)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:483:12)
	at ChildProcess.cp.emit (c:\Users\Colin\.vscode\extensions\platformio.platformio-ide-2.3.4\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1021:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
  ```
  

I had the above problem which was prevously reported.  I was advised to update to the latest versions which I have attempted to do.  However whichever of the install files I down load, whether a .exe or a .zip file, when I try to etract windows gives me a message stating that the file is corrupt.
[image]
Why does windows give this error message.  I presume that your files are not actually corrupt.
I need to get this resolved to use Platformio correctly.

Many thanks,

Colin Murdoch.

mrengineer7777, 2022-02-15:
First, make sure you have a backup image of your computer (always a good precaution).
Since you are using VSCode, go to your extensions and uninstall the Platform IDE extension.  Then install a fresh copy.  The current version is 2.4.2.  Note this is different than PIO home (3.4.1) and PIO Core (5.2.5).
Also consider updating VSCode.

ColinHMurdoch, 2022-02-15:
Thank you.
I am trying to install/upgrade VSCode as I said at the end of the call details.  However windows says that the install files are corrupt and tells me to get a fresh copy.  I included a screenshot at the end of the call for you to see.

I don’t know how to get around this problem.

Regards,

Colin Murdoch.

closed 2022-02-19 by ivankravets