platformio/platformio-vscode-ide #3054: Dependency Scanning Issue

BrandonStudio, 2022-03-08:
PlatformIO compiler does not automatically find headers referenced by dependency libraries.
In the following case, I use Adafruit SSD1306 as project dependency. Adafruit BusIO use <SPI.h> and <Wire.h>, but the compiler says they don't exist while they actually do exist.
```compiler
Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.10.13
|   |-- <Adafruit BusIO> 1.11.2
|-- <Adafruit BusIO> 1.11.2
|-- <Adafruit SSD1306> 2.5.1
|   |-- <Adafruit GFX Library> 1.10.13
|   |   |-- <Adafruit BusIO> 1.11.2
Building in release mode
Compiling .pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o
Compiling .pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_I2CDevice.cpp.o
Compiling .pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_SPIDevice.cpp.o
Compiling .pio\build\uno\libebe\Adafruit GFX Library\Adafruit_GrayOLED.cpp.o
Compiling .pio\build\uno\libebe\Adafruit GFX Library\Adafruit_SPITFT.cpp.o
Compiling .pio\build\uno\lib38d\Adafruit SSD1306\Adafruit_SSD1306.cpp.o
In file included from .pio\libdeps\uno\Adafruit BusIO/Adafruit_BusIO_Register.h:9:0,
                 from .pio\libdeps\uno\Adafruit BusIO\Adafruit_BusIO_Register.cpp:1:
.pio\libdeps\uno\Adafruit BusIO/Adafruit_I2CDevice.h:5:10: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://registry.platformio.org/search?q=header:Wire.h
*
**************************************************************

 #include <Wire.h>
          ^~~~~~~~
compilation terminated.
In file included from .pio\libdeps\uno\Adafruit BusIO\Adafruit_I2CDevice.cpp:1:0:
.pio\libdeps\uno\Adafruit BusIO\Adafruit_I2CDevice.h:5:10: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://registry.platformio.org/search?q=header:Wire.h
*
**************************************************************

 #include <Wire.h>
          ^~~~~~~~
compilation terminated.
In file included from .pio\libdeps\uno\Adafruit BusIO\Adafruit_SPIDevice.cpp:1:0:
.pio\libdeps\uno\Adafruit BusIO\Adafruit_SPIDevice.h:9:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.
*** [.pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o] Error 1
In file included from .pio\libdeps\uno\Adafruit GFX Library\Adafruit_GrayOLED.h:30:0,
                 from .pio\libdeps\uno\Adafruit GFX Library\Adafruit_GrayOLED.cpp:20:
.pio\libdeps\uno\Adafruit BusIO/Adafruit_I2CDevice.h:5:10: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://registry.platformio.org/search?q=header:Wire.h
*
**************************************************************

 #include <Wire.h>
          ^~~~~~~~
compilation terminated.
*** [.pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_I2CDevice.cpp.o] Error 1
In file included from .pio\libdeps\uno\Adafruit GFX Library\Adafruit_SPITFT.cpp:36:0:
.pio\libdeps\uno\Adafruit GFX Library\Adafruit_SPITFT.h:26:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.
*** [.pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_SPIDevice.cpp.o] Error 1
*** [.pio\build\uno\libebe\Adafruit GFX Library\Adafruit_GrayOLED.cpp.o] Error 1
*** [.pio\build\uno\libebe\Adafruit GFX Library\Adafruit_SPITFT.cpp.o] Error 1
In file included from .pio\libdeps\uno\Adafruit SSD1306\Adafruit_SSD1306.cpp:53:0:
.pio\libdeps\uno\Adafruit SSD1306\Adafruit_SSD1306.h:43:10: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://registry.platformio.org/search?q=header:SPI.h
*
*************************************************************

 #include <SPI.h>
          ^~~~~~~
compilation terminated.
*** [.pio\build\uno\lib38d\Adafruit SSD1306\Adafruit_SSD1306.cpp.o] Error 1
```
However, when I add an include.h with 
```C++
#include <SPI.h>
#include <Wire.h>
```
to include/ folder, the compilation succeeds:
```compiler
Processing uno (platform: atmelavr; board: uno; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/uno.html
PLATFORM: Atmel AVR (3.4.0) > Arduino Uno
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 31.50KB Flash   
DEBUG: Current (avr-stub) On-board (avr-stub, simavr)
PACKAGES:
 - framework-arduino-avr 5.1.0
 - toolchain-atmelavr 1.70300.191015 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.10.13
|   |-- <Adafruit BusIO> 1.11.2
|   |   |-- <Wire> 1.0
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit BusIO> 1.11.2
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit SSD1306> 2.5.1
|   |-- <Adafruit GFX Library> 1.10.13
|   |   |-- <Adafruit BusIO> 1.11.2
|   |   |   |-- <Wire> 1.0
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in release mode
Compiling .pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_BusIO_Register.cpp.o
Compiling .pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_I2CDevice.cpp.o
Compiling .pio\build\uno\lib5ad\Adafruit BusIO\Adafruit_SPIDevice.cpp.o
Compiling .pio\build\uno\libebe\Adafruit GFX Library\Adafruit_GrayOLED.cpp.o
Compiling .pio\build\uno\libebe\Adafruit GFX Library\Adafruit_SPITFT.cpp.o
Compiling .pio\build\uno\lib38d\Adafruit SSD1306\Adafruit_SSD1306.cpp.o
Archiving .pio\build\uno\lib5ad\libAdafruit BusIO.a
Archiving .pio\build\uno\libebe\libAdafruit GFX Library.a
Archiving .pio\build\uno\lib38d\libAdafruit SSD1306.a
Linking .pio\build\uno\firmware.elf
Checking size .pio\build\uno\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.4% (used 9 bytes from 2048 bytes)
Flash: [          ]   3.6% (used 1156 bytes from 32256 bytes)
```

mrengineer7777, 2022-03-11:
This is not an issue with the PIO Extension for VSCode.  I will note the dependency finder in PIO Core doesn't like circular references.  Glad you found a workaround.

ivankravets, 2022-05-28:
Duplicate of https://github.com/platformio/platformio-core/issues/3113

closed 2022-05-28 by ivankravets