platformio/platformio-vscode-ide #3058: Unable to resolve dacDisable() when compiling an ESP32/Arduino project

MrYsLab, 2022-03-09:
I am using Ubuntu 21.04 and the latest version of VSCode. In building a project for the ESP32 using Arduino for esp32doit-devkit-v1
I get the following error:
```
error: 'dacDisable' was not declared in this scope
```
The file correctly compiles when using the Arduino IDE.

The source file I used to test is:
```
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  dacWrite(25, 233);
  dacDisable(25);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

By modifying platform.ini from:
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

```
to:
```
[env:esp32doit-devkit-v1]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
board = esp32doit-devkit-v1
framework = arduino

```

It now compiles correctly.

Please update the vscode extension to correct this issue.

MrYsLab, 2022-03-11:
Thanks for the reply. Could you point to where the problem lies (which project) so I can enter an issue there?

mrengineer7777, 2022-03-11:
@MrYsLab https://github.com/platformio/platform-espressif32

I suspect your issue is because platform-espressif32 is still based on Arduino 1.0.6 and an old version (3.3.1?) of the IDF.
See https://github.com/platformio/platform-espressif32/issues/710

See also
https://github.com/espressif/arduino-esp32

I use a custom fork by TASMOTA
https://github.com/tasmota/platform-espressif32/

MrYsLab, 2022-03-11:
@mrengineer7777 Thanks.

closed 2022-03-11 by MrYsLab