platformio/platformio-vscode-ide #3063: Constant Failure in installing PlatformIO IDE on VSCode

tihsho, 2022-03-11:
During the installation of the PlatformIO IDE extension into VS Code I keep getting the following error within the "Toggle Developer Tools > Console"

platformio visual studio code [Extension Host] (node:30452) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.  (Use `Code --trace-deprecation ...` to show where the warning was created)

Figured this was an issue with VS Code as I had not used it in a while, uninstalled the VS Code instance and reinstalled with the latest download from Microsoft. Went to add the PlatformIO IDE again and I'm still getting this error. In the past it PlatformIO IDE required Python to be installed, so I went through and uninstalled the version of Python from my machine and pulled the latest 3.10.2 version from python.org and the issue persists. I'm seeing this is a fairly common issue posted back through 2018, is there no fix for this?

panxilin, 2022-04-30:
Have you solved this problem?

ndanilo8, 2022-05-24:
Facing the same issue

ivankravets, 2022-05-25:
@ndanilo8 Did you install your own Python? What is the version?

ndanilo8, 2022-05-25:
Yes I have:
Python: 3.10.4

VSCode 1.67.2
Electron: 17.4.1
Chromimum: 98.0.4758.141
NodeJS: 16.13.0
OS: Windows 10 x64 10.0.19.044

ivankravets, 2022-05-26:
Please remove `%USERPROFILE%/.platformio/penv` folder and restart VSCode.

Does it work now? Do you use any  antivirus tools?

ndanilo8, 2022-05-26:
@ivankravets 
At the time of this issue there no .platformio Folder was created even after installing the extension.

Additionally, I have fixed the issue by doing a hard reset of my machine. And now I don't have any issues whatsoever
And I don't really know how to replicate it.
Thank you so much for your time

ivankravets, 2022-06-23:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/3222

closed 2022-06-23 by ivankravets