platformio/platformio-vscode-ide #3065: Remote 'run' fails if remote agent is started by systemd daemon

maziu, 2022-03-13:
Hi,

I have an issue with executing remote run on Linux machine (Raspberry Pi) when remote agent is started by systemd daemon. 
Executed on host (my dev machine): 
```pio remote run -e rpi -t upload```

Result:
```Upon execvpe platformio ['platformio', '--force', 'run', '-d', '/home/pi/.platformio/remote/projects/vending-drv-6eb27acd96a669c471f5a8fb1beec8b271e9f48a', '-e', 'rpi', '-t', 'upload', '-t', 'nobuild', '--disable-auto-clean'] in environment id 3035755136
:Traceback (most recent call last):
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 398, in _fork
    self._execChild(path, uid, gid, executable, args, environment)
  File "/home/pi/.platformio/packages/contrib-pysite/twisted/internet/process.py", line 469, in _execChild
    os.execvpe(executable, args, environment)
  File "/usr/lib/python3.9/os.py", line 583, in execvpe
    _execvpe(file, args, env)
  File "/usr/lib/python3.9/os.py", line 616, in _execvpe
    raise last_exc
  File "/usr/lib/python3.9/os.py", line 607, in _execvpe
    exec_func(fullname, *argrest)
FileNotFoundError: [Errno 2] No such file or directory: b'/bin/platformio' 
```

Here is the systemd startup script I use:

```[Unit]
Description=PlatformIo remote agent for FW update
After=network-online.target
StartLimitBurst=3
StartLimitIntervalSec=120

[Service]
Type=simple
Restart=always
RestartSec=10
User=pi
Environment="PLATFORMIO_AUTH_TOKEN=my_token_here"
Environment="PATH=/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/home/pi/.platformio/penv/bin"
ExecStart=pio remote agent start -n "vending-1"
WorkingDirectory=/home/pi/vending/vending-drv

[Install]
WantedBy=multi-user.target
```

If I execute `pio remote agent start -n "vending-1"` directly from rpi terminal or ssh, then `pio remote run -e rpi -t upload` finishes fine.
If any more info is needed feel free to ask.

maziu, 2022-03-13:
Wow, this issue is strange. 

If I change path in the service file from 
```ExecStart=/home/pi/.local/bin/pio remote agent start -n "vending-1"
```
to 
```ExecStart=/home/pi/.platformio/penv/bin/pio remote agent start -n "vending-1"
```

Then remote run executes fine. I don't understand why actually.
My suggestion would be to add some manual how to start pio by systemd deamon in the manual if you don't plan to fix the issue. I can write the correct manual/example if you can instruct me where to add it.

closed 2022-06-04 by stale[bot]

AmrikSadhra, 2023-07-27:
Encountered this too (even with my pio located at `/home/pi/.platformio/penv/bin/pio`), ended up symlinking the pio binary into `/bin` as follows: 

`sudo ln -s /home/pi/.platformio/penv/bin/pio /bin/platformio`