platformio/platformio-vscode-ide #3081: Access denied

Roadrunner185, 2022-03-22:
I'm not able to build a project anymore after windows did an update. I'm using windows 10 and VSCode 1.65.2. It seems that this has something to do with file or folder permissions, but I don't get it fixed. Has anybody seen a similar issue? 

The following message will pop up "This app can't run on your PC".

[image]

EnricGarrigo, 2022-03-23:
Since yesterday, I have the same error with the same file 'esp32-hal-dac.c'

Roadrunner185, 2022-03-23:
I've found a solution. 
I deleted the contents of the ´.platformio/packages´, ´.platformio/platforms´, ´.platformio/.cache´ folders and rebuild the project. Afterwards it worked again.

EnricGarrigo, 2022-03-23:
Thank you so much. Solved with the same trick. After the delete, the build performs multiple 'download' and 'unpacking' I guess to update platformio.

closed 2022-06-04 by stale[bot]