platformio/platformio-vscode-ide #3085: Start PIO Home Server

edwin-agudelo, 2022-03-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.63.0
  PIO IDE: v2.4.0
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\Edwin\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\Edwin\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\Edwin\.vscode\extensions\platformio.platformio-ide-2.4.0\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-05-10:
Please update VSCode and PlatformIO IDE extension to the latest versions. Restart VSCode. Should be fixed.

Does it work now?

closed 2022-05-10 by ivankravets

edwin-agudelo, 2022-05-10:
Now it's working