platformio/platformio-vscode-ide #3090: Start PIO Home Server

bboyadzhiev, 2022-03-26:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.64.1
  PIO IDE: v2.4.1
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\<username>\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.r (c:\Users\Borislav\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (events.js:315:20)
	at ChildProcess.EventEmitter.emit (domain.js:467:12)
	at ChildProcess.cp.emit (c:\Users\Borislav\.vscode\extensions\platformio.platformio-ide-2.4.1\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:1048:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-05-10:
Please update VSCode and PlatformIO IDE extension to the latest versions. Restart VSCode. Should be fixed.

Does it work now?

closed 2022-05-10 by ivankravets

bboyadzhiev, 2022-05-13:
Sorry for the late reply!

It is OK now. Loads and builds without any issues.

Thank you!