platformio/platformio-vscode-ide #3119: After "select all" on PIO Home, cannot unselect

radonyl, 2022-04-09:
Hello, I'm using PlatformIO IDE (v2.4.3) on Visual Studio Code (Version: 1.66.0)

When I enter PIO Home and use cmd + A shortcut to "select all", and whole user interface becomes white.

[image]

russwills, 2022-05-01:
Same problem.
VS Code: Version: 1.66.2 (Universal) Commit: 
PlatformIO Core: 5.2.5
PlatformIO Home: 3.4.1
Once I use Cmd-A (habit) I can't find a way to deselect the interface without closing the PIO Home tab and reopening.

SLboat, 2023-01-06:
[image]

it's happen here too

PR #3451 (2023-02-24): Fix select all, undo, and redo on macOS
`selectAll`, `undo`, and `redo` are not supported events in `document.addEventListener` (https://developer.mozilla.org/en-US/docs/Web/Events).

To handle select all, we can detect a `selectStart` event, which only fires if something outside the iframe is selected. This is only achievable if select all is used as the iframe takes up the entire window. If so, then we execute the command and prevent the incorrect selection from occurring.

Undo/redo events do not exist (https://stackoverflow.com/a/15516286), so we can instead manually listen for Cmd+Z/Cmd+S [... cut]
closed 2023-02-24 by ivankravets