platformio/platformio-vscode-ide #3122: Device Monitor is not working

osamahnini, 2022-04-11:
I try to use device monitor in PlatformIO IDE in VSCode and I got this error:
Error: Traceback (most recent call last):
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\commands\device\command.py", line 189, in device_monitor  
    device_helpers.register_filters(platform=platform, options=kwargs)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\commands\device\helpers.py", line 117, in register_filters    load_monitor_filters(
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\commands\device\helpers.py", line 105, in load_monitor_filters
    load_monitor_filter(path, options)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\commands\device\helpers.py", line 83, in load_monitor_filter
    module = load_python_module("platformio.commands.device.filters.%s" % name, path)
  File "C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\compat.py", line 62, in load_python_module
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\User-HP\.platformio\platforms\espressif32\monitor\filter_exception_decoder.py", line 20, in <module>
    from platformio.compat import path_to_unicode, WINDOWS, PY2
ImportError: cannot import name 'PY2' from 'platformio.compat' (C:\Users\User-HP\.platformio\penv\lib\site-packages\platformio\compat.py)

Fa-b, 2022-04-12:
Exact same issue here.
`pio system info` yields:
```
--------------------------  -------------------------------------------------------------
PlatformIO Core             5.3.0b5
Python                      3.10.2-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\Fa-b\.platformio
PlatformIO Core Executable  C:\Users\Fa-b\.platformio\penv\Scripts\platformio.exe
Python Executable           c:\users\fa-b\.platformio\penv\scripts\python.exe
Global Libraries            0
Development Platforms       2
Tools & Toolchains          10
--------------------------  -------------------------------------------------------------
```

Fa-b, 2022-04-12:
[this](https://github.com/platformio/platform-espressif32/issues/759) solution works for `platform-espressif8266` too. The lines are `21`, `258`, and `266` in that case.

closed 2022-06-23 by ivankravets