platformio/platformio-vscode-ide #3123: PlatformIO IDE on code-server running in a Docker container

silvio-vallorani, 2022-04-12:
Hi guys, this is probably not the right place to ask this question, but I don't know where to start such a thread ... don't have a Discord server or a Slack workspace or a Team on Microsoft Teams?

Anyway, has anyone tried running PlatformIO IDE inside a code-server instance running in a docker container like this https://hub.docker.com/r/codercom/code-server ?

I have tried and made a good point, but there are several obstacles that I don't know how to deal with ...

Thank you for your nice work!

thedanbob, 2022-04-18:
I have it mostly working using the [linuxserver docker image](https://hub.docker.com/r/linuxserver/code-server). Here's a snippet of my docker compose file:

```yaml
version: '2.4'
services:
  code-server:
    image: lscr.io/linuxserver/code-server
    restart: always
    environment:
      - TZ=America/Chicago
      - PUID=1000
      - PGID=1000
      - DOCKER_MODS=linuxserver/mods:code-server-python3
      - 'EXTENSIONS_GALLERY={"serviceUrl":"https://marketplace.visualstudio.com/_apis/public/gallery","itemUrl":"https://marketplace.visualstudio.com/items"}'
    volumes:
      - /srv/docker/code-server:/config
      - /projects/path/on/host:/projects
```

The `pio` command works in the built-in terminal but the embedded PlatformIO web pages don't work. And unfortunately it's not possible to get them working currently as the extension assumes you're accessing it from the same host it's running on.

closed 2022-09-20 by stale[bot]

melwinek, 2023-05-24:
Has anyone been able to get PlatformIO running in docker in code-server?