platformio/platformio-vscode-ide #3129: `first parameter of allocation function must be of type 'size_t'C/C++(351)`

AgainPsychoX, 2022-04-15:
When using PlatformIO with , I get `first parameter of allocation function must be of type 'size_t'C/C++(351)` error (and other related to placement `new` usage).

Project details (from `platformio.ini`):
```
platform = espressif8266
board = nodemcuv2
framework = arduino
```

A lot of issues in `new` header:
```
C:\Users\PsychoX\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\new
```
[image]

When using placement `new` operator:
[image]

**Please note that compilation works** without any problems (with no warnings). 

It might be issue with C/Cpp Tools extension which is used by PlatformIO (by settings `c_cpp_preferences.json` file).

ahmadfarisfs, 2022-05-22:
[image]
I can confirm that I also had issues with this

ivankravets, 2022-06-23:
Please forward to https://github.com/microsoft/vscode-cpptools/issues