platformio/platformio-vscode-ide #3137: Start PIO Home Server

cjogo, 2022-04-22:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.66.2
  PIO IDE: v2.4.3
  System: Linux, 5.13.0-40-generic, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 121, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/commands/home/command.py", line 88, in cli
    run_server(
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/commands/home/run.py", line 63, in run_server
    contrib_dir = get_core_package_dir("contrib-piohome")
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/core.py", line 48, in get_core_package_dir
    pkg = pm.get_package(spec)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/base.py", line 236, in get_package
    for pkg in self.get_installed():
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/package/manager/base.py", line 209, in get_installed
    pkg = PackageItem(pkg_dir)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/package/meta.py", line 405, in __init__
    self.metadata = self.load_meta()
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/package/meta.py", line 440, in load_meta
    return PackageMetaData.load(manifest_path)
  File "/home/comp/.platformio/penv/lib/python3.8/site-packages/platformio/package/meta.py", line 393, in load
    data["spec"] = PackageSpec(**data["spec"])
TypeError: __init__() got an unexpected keyword argument 'uri'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at /home/comp/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:9851
	at ChildProcess.c (/home/comp/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Socket.<anonymous> (node:internal/child_process:450:11)
	at Socket.emit (node:events:390:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

cjogo, 2022-04-22:
**************************************************************************************************************************************
Obsolete PIO Core v5.2.5 is used (previous was 6.0.0a1)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
**************************************************************************************************************************************
Processing pico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: pico; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------

ivankravets, 2022-04-25:
Please upgrade VSCode and PlatformIO IDE extension for the VSCode to the latest versions. Restart VSCode.

ivankravets, 2022-04-25:
Run `pio upgrade --dev`. Restart VSCode.

closed 2022-04-25 by ivankravets

danilomendonca, 2022-04-26:
Hi there,

I am facing the same issue. I was attempting to start native unit tests in debug mode, so I have upgraded to the unstable PIO Core (6.0.0a1). Once I have concluded these tests (without success), I rollbacked to the stable version following the instructions on the Platformio CLI installation page.

After successfully returning to the stable version (PIO Core 5.2.5), I started seeing the same error mentioned in this thread: 

`TypeError: __init__() got an unexpected keyword argument 'uri'`

Also intriguing, I am seeing the following warning:

`Obsolete PIO Core v5.2.5 is used (previous was 6.0.0a1)`

which makes no sense as I have uninstalled the unstable version. To be sure, I have double-checked that a single installation method was in place -- I have removed pip-based installation in favor of the virtual environment approach (installed through PIO scripts). The warning message is still there even if I switch methods -- removing any reference from PATH to the virtual env installation and installing PIO Core through pip.

Using the unstable PIO Core is not an option for my safety-critical software, hence I wonder what is preventing me from using the stable version -- are there any python scripts that got updated and not rollbacked?

As a temporary solution, I have added "uri" to the list of expected arguments of the `PackageSpec` in meta.py. I couldn't figure out how this argument ended up in the constructor list -- if I print the list of arguments for each package before calling the constructor, uri does not show up for the first package that breaks the script (toolchain-gccarmnoneeabi), nor it is present in .piopm file from that same package folder.

Any support on this will be appreciated. 

Regards,
Danilo

ivankravets, 2022-04-26:
@danilomendonca  - just remove `~/.platformio` folder and restart IDE.

Did you have any problems with PlatformIO Core 6.0?

danilomendonca, 2022-04-26:
@ivankravets thanks.

To be honest I did not know there was a new major available. We tend to be conservative regarding upgrades in our stack due to the critical nature of our product. My intent to use the edge version was to experiment with debugging in native unit tests.

I will take some time to study the notes and migration steps in a sandbox environment. If everything goes well, I will suggest the latest stable version to the team.