platformio/platformio-vscode-ide #3149: Start PIO Home Server

henrikschwarz, 2022-04-29:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.66.2
  PIO IDE: v2.4.3
  System: Linux, 5.17.4-arch1-1, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/platformio/__main__.py", line 121, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/platformio/commands/home/command.py", line 88, in cli
    run_server(
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/platformio/commands/home/run.py", line 85, in run_server
    uvicorn.run(
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/main.py", line 463, in run
    server.run()
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/server.py", line 60, in run
    return asyncio.run(self.serve(sockets=sockets))
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/server.py", line 67, in serve
    config.load()
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/config.py", line 450, in load
    ws_protocol_class = import_from_string(WS_PROTOCOLS[self.ws])
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/importer.py", line 24, in import_from_string
    raise exc from None
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/importer.py", line 21, in import_from_string
    module = importlib.import_module(module_str)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/protocols/websockets/auto.py", line 17, in <module>
    from uvicorn.protocols.websockets.websockets_impl import WebSocketProtocol
  File "/home/henriko/.platformio/penv/lib/python3.10/site-packages/uvicorn/protocols/websockets/websockets_impl.py", line 7, in <module>
    from websockets.extensions.permessage_deflate import ServerPerMessageDeflateFactory
ModuleNotFoundError: No module named 'websockets.extensions'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

    at cmd (/home/henriko/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/home.js:195:11)
    at ChildProcess.c (/home/henriko/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:177:7)
    at ChildProcess.emit (node:events:390:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1064:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

henrikschwarz, 2022-04-29:
Fixed it by activating `~/.platformio/penv/bin/activate` and installing websockets with `pip install websockets`.

rikilshah, 2022-05-03:
> Fixed it by activating `~/.platformio/penv/bin/activate` and installing websockets with `pip install websockets`.

I am inexperienced in Core VSCODE commands. Will you be able to provide slightly detailed instructions?

henrikschwarz, 2022-05-03:
The issue arises from the python virtual environment that PlatformIO uses. The virtual environment doesn't have the python package `websockets`. So I had to manually activate the virtual environment and install the package. 

I don't know if there is a VSCode command to fix it. I used my terminal and did it. If you are on Linux then you can type this as a temporary solution:
- `source ~/.platformio/penv/bin/activate`
- `pip install websockets`

rikilshah, 2022-05-03:
I don't have bin folder at that exact location. I tried reinstalling PlatformIO to avail no result. Am I missing something?

rikilshah, 2022-05-03:
> I don't have bin folder at that exact location. I tried reinstalling PlatformIO to avail no result. Am I missing something?
> 
> PS: I am using Windows 10.

Okay, So far I made progress as per following:

1. in ~/.platformio/penv/Script folder, "Activate" script executed in Command Shell which activated Python environment.
2. Installed websockets. 
3. Restarted VSCode
4. Same error.

henrikschwarz, 2022-05-03:
Can you post your errors?

ishay320, 2022-05-04:
I'm using windows, and I did what @henrikschwarz said and it didn't worked, so I installed `websockets` on my computer and it worked.
I don't know if the installing it on the main python fixes it or it needs both..