platformio/platformio-vscode-ide #3150: How do I completely remove the PlatformIO Extension from VSCode?

cquick01, 2022-05-02:
I've uninstalled the extension through VSCode, and scoured my machine for any reference to "platformio", and VSCode still shows the PIO shortcuts in the right click menu ("PlatformIO: Build", "PlatformIO: Upload", "PlatformIO: Open Serial Monitor").

How do I get rid of PlatformIO and why doesn't uninstalling the extension normally remove everything?

ameerhazo, 2022-05-03:
try deleting the .platformio file on your user profile in C:/Users/your_username

cquick01, 2022-05-03:
Nope, that didn't work. It did delete over a half gig of stuff though

ivankravets, 2022-06-23:
This is not our command. Please remove other extensions that add these commands to the context menu.

closed 2022-06-23 by ivankravets