platformio/platformio-vscode-ide #3151: Start PIO Home Server

LEDZB, 2022-05-03:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.66.2
  PIO IDE: v2.4.3
  System: Windows_NT, 10.0.19043, x64

  %23 Exception
  ```
  Error: ERROR:    [Errno 10048] error while attempting to bind on address ('127.0.0.1', 8010): ͨ��ÿ���׽��ֵ�ַ(Э��/�����ַ/�˿�)ֻ����ʹ��һ�Ρ�

	at c:\Users\DZB\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.c (c:\Users\DZB\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\DZB\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-05-17:
Do you use any firewall?

LEDZB, 2022-05-18:
Hello, I don't use firewall. Is this related to firewall？

ivankravets, 2022-05-18:
and antivirus tools?

LEDZB, 2022-05-18:
> and antivirus tools?
No, I just use the Windows Defender that comes with win10. What's wrong with this

ivankravets, 2022-05-18:
Please open [PlatformIO Core CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type 

```
pio home
```

It should be opened in the browser. Do you see any output?

ivankravets, 2022-06-23:
Please reopen if you still need help

closed 2022-06-23 by ivankravets

LEDZB, 2022-06-23:
> Please reopen if you still need help

This problem has been solved some time ago. Thank you very much for your help,