platformio/platformio-vscode-ide #3155: Start PIO Home Server

LJC-use, 2022-05-06:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.0
  PIO IDE: v2.4.3
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\platformio\__main__.py", line 121, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 88, in cli
    run_server(
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\platformio\commands\home\run.py", line 85, in run_server
    uvicorn.run(
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\uvicorn\main.py", line 463, in run
    server.run()
  File "C:\Users\ljc19\.platformio\penv\lib\site-packages\uvicorn\server.py", line 60, in run
    return asyncio.run(self.serve(sockets=sockets))
  File "D:\python\lib\asyncio\runners.py", line 39, in run
    loop = events.new_event_loop()
  File "D:\python\lib\asyncio\events.py", line 782, in new_event_loop
    return get_event_loop_policy().new_event_loop()
  File "D:\python\lib\asyncio\events.py", line 673, in new_event_loop
    return self._loop_factory()
  File "D:\python\lib\asyncio\windows_events.py", line 315, in __init__
    super().__init__(proactor)
  File "D:\python\lib\asyncio\proactor_events.py", line 632, in __init__
    self._make_self_pipe()
  File "D:\python\lib\asyncio\proactor_events.py", line 763, in _make_self_pipe
    self._ssock, self._csock = socket.socketpair()
  File "D:\python\lib\socket.py", line 644, in socketpair
    ssock, _ = lsock.accept()
  File "D:\python\lib\socket.py", line 293, in accept
    fd, addr = self._accept()
TimeoutError: timed out

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

C:\Users\ljc19\.platformio\penv\lib\site-packages\platformio\__main__.py:153: RuntimeWarning: coroutine 'Server.serve' was never awaited
  exit_code = int(str(e)) if str(e).isdigit() else 1
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
Exception ignored in: <function BaseEventLoop.__del__ at 0x000001FF6BFBA5F0>
Traceback (most recent call last):
  File "D:\python\lib\asyncio\base_events.py", line 687, in __del__
    self.close()
  File "D:\python\lib\asyncio\proactor_events.py", line 687, in close
    self._close_self_pipe()
  File "D:\python\lib\asyncio\proactor_events.py", line 755, in _close_self_pipe
    self._ssock.close()
AttributeError: 'ProactorEventLoop' object has no attribute '_ssock'

	at c:\Users\ljc19\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.c (c:\Users\ljc19\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\ljc19\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-05-10:
What is your custom Python version?

LJC-use, 2022-05-10:
Thank you. I fixed the problem. My firewall blocked it-------- 原始邮件 --------发件人： Ivan Kravets ***@***.***>日期： 2022年5月10日周二 晚上8:55收件人： platformio/platformio-vscode-ide ***@***.***>抄送： LJC-use ***@***.***>, Author ***@***.***>主    题： Re: [platformio/platformio-vscode-ide] Start PIO Home Server (Issue #3155)
What is your custom Python version?

ivankravets, 2022-05-10:
Happy coding with PlatformIO!