platformio/platformio-vscode-ide #3172: Start PIO Home Server

mtrivelli, 2022-05-12:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.1
  PIO IDE: v2.4.3
  System: Windows_NT, 10.0.19044, x64

  %23 Exception
  ```
  Error: Usage: python -m platformio [OPTIONS] COMMAND [ARGS]...
Try 'python -m platformio -h' for help.

Error: No such command "home"

	at c:\Users\Manuel\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.c (c:\Users\Manuel\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\Manuel\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-05-12:
Could you reproduce this issue after the VSCode restart?

ivankravets, 2022-05-17:
Please update VSCode and PlatformIO IDE extension to the latest versions. Restart VSCode.

closed 2022-05-17 by ivankravets