platformio/platformio-vscode-ide #3181: Cannot compile Marlin all of a sudden

HGFTek, 2022-05-16:
Since this morning, not a single of my computers can compile Marlin. Normal Arduino code still works fine.
all computers give the following warnings, and eventually returns with a compilation failed error.
the issue has been raised in the VSCode github, but it has been closed as they deem it a PlatformIO issue.

here are the warnings appearing in the terminal window.
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead

jbaiao, 2022-05-16:
Me too!
Two machine and nothing works...

coldgap, 2022-05-16:
Me three.

Forunnerod, 2022-05-17:
Me Four

KevinSchrama, 2022-05-17:
I had the same issue, this thread and comment helped getting it working again: https://github.com/platformio/platformio-core/issues/4270#issuecomment-1128100663

SikVac, 2022-05-17:
Platformio 6 is on beta and incompatible with Marlin ! Go back to 5.2.5 !
[image]
pip install platformio==5.2.5

ivankravets, 2022-05-17:
Duplicate of https://github.com/platformio/platformio-core/issues/4270

closed 2022-05-17 by ivankravets