platformio/platformio-vscode-ide #3182: Cannot remove temporary directory /home/..../.../.pio/build

dougparkarosa, 2022-05-17:
I'm using Ubuntu 22.04. VSCode 1.67.1. Upgraded to Core: 6.0.0/3.4.1.

I get "Can not remove temporary directory `/home/doug/PHD/SNBPL_Mesh/.pio/build`. Please remove it manually to avoid build issues" 
Followed by:
Error: Traceback (most recent call last):
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super().invoke(ctx)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/commands/run/command.py", line 142, in cli
    process_env(
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/commands/run/command.py", line 184, in process_env
    print_processing_header(name, config, verbose)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/commands/run/command.py", line 218, in print_processing_header
    for k, v in config.items(env=env):
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 224, in items
    return [(option, self.get(section, option)) for option in self.options(section)]
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 224, in <listcomp>
    return [(option, self.get(section, option)) for option in self.options(section)]
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 341, in get
    value = self.getraw(section, option, default)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 242, in getraw
    return self._getraw(section, option, default)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 303, in _getraw
    return self._expand_interpolations(section, value)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 312, in _expand_interpolations
    return self.VARTPL_RE.sub(
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 313, in <lambda>
    lambda match: self._re_interpolation_handler(parent_section, match), value
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 329, in _re_interpolation_handler
    value = self.get(section, option)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 341, in get
    value = self.getraw(section, option, default)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 242, in getraw
    return self._getraw(section, option, default)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 303, in _getraw
    return self._expand_interpolations(section, value)
  File "/home/doug/.platformio/penv/lib/python3.10/site-packages/platformio/project/config.py", line 312, in _expand_interpolations
    return self.VARTPL_RE.sub(
TypeError: sequence item 8: expected str instance, int found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process "platformio 'run', '--environment', 'ttgo-t-beam'" terminated with exit code: 1.

I get this when I do clean or build. 

I have confirmed that the .pio/build directory is not there. I was able to get better results creating a new blank project. I've tried reinstalling. Tried deleting .pio directory as well. 

I'm not certain that the upgrade is the cause as I haven't worked on this code for a few weeks. I also recently upgraded to Ubuntu 22.04. For awhile I was thinking that Python 3.10 might be the issue.

ivankravets, 2022-05-17:
Duplicate of https://github.com/platformio/platformio-core/issues/4271

Could you share your `platformio.ini` contents?

closed 2022-05-17 by ivankravets

dougparkarosa, 2022-05-17:
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = ttgo-t-beam

[warnings]
src = -Wall -Wextra
lib = -Wno-unused-function -Wno-unused-variable

[defines]
src = 
lib = 

[etl]
defines = -D PROFILE_GCC_GENERIC -D ETL_NO_STL

[lmic]
defines = -D LMIC_LORAWAN_SPEC_VERSION=LMIC_LORAWAN_SPEC_VERSION_1_0_3 -D CFG_us915 -D DISABLE_INVERT_IQ_ON_RX

[env]
framework = arduino
debug_tool = esp-prog
platform = espressif32
monitor_speed = 115200
check_tool = 
	cppcheck
	clangtidy
check_skip_packages = yes
check_flags = 
	clangtidy: --checks=-*,cert-*,clang-analyzer-*,bugprone-*,hicpp-*,modernize-*,-modernize-use-trailing-return-type,performance-*,portability-*,readability-*,llvm-*,misc-*,performance-*,portability-*
	cppcheck: --suppress=*:*/.pio/* --inline-suppr
lib_extra_dirs = ../Libraries
lib_ldf_mode = deep+
build_flags = ${warnings.lib} ${lmic.defines} ${etl.defines} ${defines.lib} -D MONITOR_SPEED=${env.monitor_speed}
build_src_flags = {$warnings.src} ${lmic.defines}  ${etl.defines} ${defines.src} -D MONITOR_SPEED=${env.monitor_speed}

[env:heltec_wifi_lora_32_V2]
board = heltec_wifi_lora_32_V2
build_flags = ${env.build_flags} -D HELTEC_WIFI_LORA -D USE_U8G2=1 -D ARDUINO_ARCH_ESP32
lib_deps = 
	adafruit/Adafruit SSD1306@^2.5.0
	adafruit/Adafruit BusIO@^1.9.9

[env:ttgo-t-beam]
platform = espressif32
board = ttgo-t-beam
build_flags = ${env.build_flags} -D TTGO_T_BEAM -D USE_ADAFRUIT=1 -D ARDUINO_ARCH_ESP32
lib_deps = 
	adafruit/Adafruit SSD1306 @ ^2.5.1
	adafruit/Adafruit BusIO@^1.9.9

dougparkarosa, 2022-05-17:
This much more terse platformio.ini also reproduces the problem.

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env]
framework = arduino
platform = espressif32
monitor_speed = 115200
build_flags = -D MONITOR_SPEED=115200
board = ttgo-t-beam

ivankravets, 2022-05-17:
Fixed in the latest release. Please upgrade PIO Core https://community.platformio.org/t/platformio-core-6-0/27730/4?u=ivankravets