platformio/platformio-vscode-ide #3187: Start PIO Home Server

milindkhurd, 2022-05-18:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.1
  PIO IDE: v2.4.3
  System: Darwin, 21.3.0, arm64

  %23 Exception
  ```
  Error: Error: HTTPSConnectionPool(host='api.registry.nm1.platformio.org', port=443): Max retries exceeded with url: /v3/packages/platformio/tool/contrib-piohome (Caused by SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: self signed certificate in certificate chain (_ssl.c:997)')))

    at cmd (/Users/mkhurd/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/home.js:195:11)
    at ChildProcess.c (/Users/mkhurd/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/webpack:/platformio-node-helpers/src/proc.js:177:7)
    at ChildProcess.emit (node:events:390:28)
    at ChildProcess.emit (node:domain:475:12)
    at maybeClose (node:internal/child_process:1064:16)
    at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-06-23:
Please reopen if you still need help

closed 2022-06-23 by ivankravets

milindkhurd, 2022-06-28:
Sorry had an urgent matter to deal with, I did not install Python manually.

ivankravets, 2022-06-29:
Could you provide a remote session to reproduce this issue? Please contact me@ikravets.com.