platformio/platformio-vscode-ide #3190: Start PIO Home Server

bsbtgit, 2022-05-20:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.1
  PIO IDE: v2.4.3
  System: Linux, 5.13.0-41-generic, x64

  %23 Exception
  ```
  Error: ERROR:    [Errno 98] error while attempting to bind on address ('127.0.0.1', 8010): address already in use

	at /home/lenovo/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:9851
	at ChildProcess.c (/home/lenovo/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

ivankravets, 2022-06-23:
Have you resolved this issue?

bsbtgit, 2022-06-23:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。