platformio/platformio-vscode-ide #3195: Vscode Update - Platform IO not working

fabltd, 2022-05-22:
Hi 

After updating VS code to the latest version I get the following error when using the UI buttons:

command 'platformio-ide.build' not found

If I select a termaial and use the PIO CLI commands platform IO works as expected.

PIO Version: PlatformIO Core, version 6.0.1

VSCODE Version: 1.67.2

Thanks

fabltd, 2022-05-22:
Update it seems to be related to only some platform IO projects.

ivankravets, 2022-05-22:
Have you read the migration guide? See https://docs.platformio.org/en/latest/core/migration.html

if you still experience issues, please forward to https://community.platformio.org/

fabltd, 2022-05-22:
Why should I need a migration guide. This was working pefectly until the latest VSCode update.

Now Platform IO is stuck loading tasks and rebuilding intellisense.

fabltd, 2022-05-22:
Error: Multiple requests to rebuild the project "Fabcontrols-firmware" index have been received!
Automatic index rebuilding process has been terminated for 10 minutes.
	at A.value (/Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:790605)
	at A.value (/Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:792714)
	at A.value (/Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:794550)
	at /Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/inde...

ivankravets, 2022-05-22:
Do you have any extra scripts that manipulate with packages? Please share your PlatformIO.ini file. Thanks

fabltd, 2022-05-22:
Hi 

Yes I do. It was working.

Is there a way to debug it?

Platformio.ini loads:

`extra_scripts = pre:./preInstallDependencies.py `

preinstallDependencies has the following:
```

Import("env")
import sys
sys.path.insert(1, './build')

import main

try:
    main.request_eui()

except ImportError:
    env.Execute("$PYTHONEXE -m pip install some_package")
    env.Execute("$PYTHONEXE -m pip install --upgrade pip --trusted-host pypi.org --trusted-host files.pythonhosted.org")
    env.Execute("$PYTHONEXE -m pip install firebase-admin requests")
    env.Execute("$PYTHONEXE -m pip install qrcode reportlab")
    env.Execute("$PYTHONEXE -m pip install requests ")
```

main.py then loads a bunch of other scripts. 

These all work but seem to break intellisense. Any way to exclude them?

J

fabltd, 2022-05-22:
OK 

This is what breaks it:

main.py

def request_eui():
   print("Scan Device EUI: ")
   eui = input()

Requesting for user input breaks it. A Intelliesense is stuck in a loop.

I followed an example you sent me for this, 

Never the less since VScode update it now breaks platform io.

Jeremy

fabltd, 2022-05-22:
Simply adding these two lines to the main extrascripts.py breaks it:
```

print("Scan Device EUI: ")
eui = input()

```

ivankravets, 2022-05-23:
It should not work even with 5.x version. You need avoid STDIN in extra scripts when `idedata` request comes. 

The docs describes this https://docs.platformio.org/scripting/actions.html

closed 2022-05-23 by ivankravets

fabltd, 2022-05-23:
Hi

That link does not work?

Followed this link you sent me originally.

https://community.platformio.org/t/user-input-in-build-script/2650/10

JT

ivankravets, 2022-05-23:
Please check docs regarding pre/extra scripts. I remember we described this case a few years ago. 

Sorry, don’t have normal access to the docs.

fabltd, 2022-05-23:
The only thing I can find is the following:

```
# (Optional) Do not run extra script when IDE fetches C/C++ project metadata
#
from SCons.Script import COMMAND_LINE_TARGETS

if "idedata" in COMMAND_LINE_TARGETS:
    env.Exit(0)
```

This appears to let platform IO start but it is still Rebuilding Intellisense. Is there some way to debug this?

fabltd, 2022-05-25:
Any fix for this ? The docs do not explain this well enough. I cannot request user input as suggested.

ivankravets, 2022-05-27:
1. Please upgrade to the latest PIO Core (`pio upgrade --dev`)
2. Update your extra scripting according to the updated example https://docs.platformio.org/en/latest/scripting/examples/asking_for_input.html

Does it work now?

fabltd, 2022-05-27:
Hi Ivan

Yes Platform IO now starts and Intellitesene stops but its reporting an error after some time:

Error: Multiple requests to rebuild the project "Fabcontrols-firmware" index have been received!
Automatic index rebuilding process has been terminated for 10 minutes.
	at A.value (/Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:790605)
	at A.value (/Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:792714)
	at A.value (/Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:794550)
	at /Users/jeremy/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/inde...

Thanks

fabltd, 2022-05-31:
Hi Ivan

Any update on the error above?

ivankravets, 2022-05-31:
Hi Jeremy,

Could you try these instructions https://github.com/platformio/platformio-vscode-ide/issues/3207#issuecomment-1140519626 ?