platformio/platformio-vscode-ide #3204: Start PIO Home Server

cwrigh13, 2022-05-27:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.2
  PIO IDE: v2.4.3
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  %23 pylint: disable=no-value-for-parameter
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super().invoke(ctx)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\click\core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\commands\home\command.py", line 88, in cli
    run_server(
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\commands\home\run.py", line 63, in run_server
    contrib_dir = get_core_package_dir("contrib-piohome")
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\package\manager\core.py", line 53, in get_core_package_dir
    assert pm.install(spec)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48, in install
    pkg = self._install(spec, skip_dependencies=skip_dependencies, force=force)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 100, in _install
    pkg = self.install_from_registry(spec, search_qualifiers)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\package\manager\_registry.py", line 111, in install_from_registry
    package = self.fetch_registry_package(spec)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\package\manager\_registry.py", line 175, in fetch_registry_package
    result = regclient.get_package(self.pkg_type, spec.owner, spec.name)
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\clients\registry.py", line 147, in get_package
    return self.fetch_json_data(
  File "C:\Users\cwrig\.platformio\penv\lib\site-packages\platformio\clients\http.py", line 153, in fetch_json_data
    return json.loads(result)
  File "C:\Users\cwrig\.platformio\python3\lib\json\__init__.py", line 346, in loads
    return _default_decoder.decode(s)
  File "C:\Users\cwrig\.platformio\python3\lib\json\decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "C:\Users\cwrig\.platformio\python3\lib\json\decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

	at c:\Users\cwrig\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:9851
	at ChildProcess.c (c:\Users\cwrig\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\platformio-node-helpers\dist\index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at ChildProcess.cp.emit (c:\Users\cwrig\.vscode\extensions\platformio.platformio-ide-2.4.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (node:internal/child_process:1064:16)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)
  ```

cwrigh13, 2022-05-27:
hi any idea how i can get this to work? (i am an idiot)

ivankravets, 2022-05-27:
Proxy? Internet?

closed 2022-06-23 by ivankravets