platformio/platformio-vscode-ide #3205: library manager install for Azure SDK for C adds backwards compatibility that causes errors

critchards, 2022-05-29:
When using the library manager to install Azure SDK for C, lib_deps gets `azure/Azure SDK for C@^1.0.0-beta.4` which throws an error on build of `Error: Could not find the package with 'azure/Azure SDK for C @ ^1.0.0-beta.4' requirements for your system 'windows_amd64'`

library manager should include specific version as `azure/Azure SDK for C@1.0.0-beta.4`

See [thread for discussion](https://community.platformio.org/t/azure-sdk-for-c-1-0-0-beta-4-cannot-find-package-for-your-system-windows-amd64/27990)

ivankravets, 2022-05-29:
Where did you find this instruction? It is not possible to use operators for the build versions. See installation instructions https://registry.platformio.org/libraries/azure/Azure%20SDK%20for%20C/installation

We use this Python library https://github.com/rbarrois/python-semanticversion

closed 2022-05-29 by ivankravets

maxgerhardt, 2022-05-29:
The *library manager* adds it in automatically, causing the failure

[image]

[image]