platformio/platformio-vscode-ide #3206: Start PIO Home Server

RistoAP, 2022-05-29:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.67.2
  PIO IDE: v2.4.3
  System: Darwin, 15.6.0, x64

  %23 Exception
  ```
  Error: Error: You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

	at /Users/ripirhon/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:9851
	at ChildProcess.c (/Users/ripirhon/.vscode/extensions/platformio.platformio-ide-2.4.3/node_modules/platformio-node-helpers/dist/index.js:1:5134)
	at ChildProcess.emit (node:events:390:28)
	at ChildProcess.emit (node:domain:475:12)
	at maybeClose (node:internal/child_process:1064:16)
	at Socket.<anonymous> (node:internal/child_process:450:11)
	at Socket.emit (node:events:390:28)
	at Socket.emit (node:domain:475:12)
	at Pipe.<anonymous> (node:net:687:12)
  ```

ivankravets, 2022-05-29:
> You are not connected to the Internet.
PlatformIO needs the Internet connection to download dependent packages or to work with PlatformIO Account.

Please check the Internet connection and restart VSCode.

closed 2022-05-29 by ivankravets

RistoAP, 2022-05-30:
HiI am connected to internet. This issue report is clear indication of it. I have restarted Code several times and my Mac also.There were certificate issue as I have old Mac, but I installed new certs and at least Safari and Chrome are using new Certs fit SSL so I assume keychain works.