platformio/platformio-vscode-ide #3219: "Update All" command calls in invalid command

maxgerhardt, 2022-06-09:
https://github.com/platformio/platformio-vscode-ide/blob/4c0e720e08d8558f8a48e33d63c444198dfbe0e1/package.json#L176-L180

https://github.com/platformio/platformio-vscode-ide/blob/fa82c0dc158e6794059d038dae059fd390cd6995/src/main.js#L264-L266

But this leads to

```
This command is deprecated and will be removed in the next releases.
Please use pio pkg update instead.
```

See https://community.platformio.org/t/update-all-in-visual-studio-code/28246

maxgerhardt, 2022-06-14:
Library installs also exhibit a similiar warning per [community post](https://community.platformio.org/t/vscode-this-command-is-deprecated/28316?u=maxgerhardt).

ivankravets, 2022-06-14:
Thanks! We work on the updates to the PIO Home. Also, the patch version for VSCode's extension is planned too.

commit 956fc9cf (2022-06-23): Remove deprecated "updates" related command in favor of project dependency management  // Resolve #3219
closed 2022-06-23 by ivankravets