platformio/platformio-vscode-ide #3222: Platform IDE will not install.

plug-it-in, 2022-06-21:
I have tried the solutions in Issue 221 and it still will not install. This is the command prompt execution of the installer

C:\.platformio\.cache\tmp>python get-platformio-1.1.1.py
Installer version: 1.1.1
Platform: Windows-10
Python version: 3.10.5 (tags/v3.10.5:f377153, Jun  6 2022, 16:14:13) [MSC v.1929 64 bit (AMD64)]
Python path: C:\Users\Pete Rawlings\AppData\Local\Programs\Python\Python310\python.exe
Creating a virtual environment at C:\.platformio\penv
Traceback (most recent call last):
  File "C:\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 69, in <module>
    main()
  File "C:\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 61, in main
    bootstrap()
  File "C:\.platformio\.cache\tmp\get-platformio-1.1.1.py", line 47, in bootstrap
    pioinstaller.__main__.main()
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\pioinstaller\__main__.py", line 133, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\click\core.py", line 1128, in __call__
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\click\core.py", line 1053, in main
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\click\core.py", line 1637, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\click\core.py", line 1395, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\click\core.py", line 754, in invoke
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\click\decorators.py", line 26, in new_func
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\pioinstaller\__main__.py", line 61, in cli
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\pioinstaller\core.py", line 88, in install_platformio_core
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\pioinstaller\core.py", line 70, in install_platformio_core
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\pioinstaller\core.py", line 98, in _install_platformio_core
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\pioinstaller\penv.py", line 71, in create_core_penv
  File "C:\.platformio\.cache\tmp\.piocore-installer-ouwlk0ph\tmp4c0pambo\pioinstaller.zip\pioinstaller\penv.py", line 141, in init_state
  File "C:\Users\Pete Rawlings\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\Pete Rawlings\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 524, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['C:\\.platformio\\penv\\Scripts\\python.exe', '-c', "import sys; version=sys.version_info; print('%d.%d.%d'%(version[0],version[1],version[2]))"]' returned non-zero exit status 101.

C:\.platformio\.cache\tmp>

plug-it-in, 2022-06-21:
Heres the console log as wel
[vscode-app-1655827161892.log](https://github.com/platformio/platformio-vscode-ide/files/8950533/vscode-app-1655827161892.log)
l

plug-it-in, 2022-06-21:
This is VERY annoying. Tried on vanilla VMWARE image and all worked fine. So I know its something on my machine but no idea what. So though I'd go with Stand Alone VS Code. Fired it up and the installer doesn't use the standalone environment and still tries to operate in the User space. How do you install the Platformio IDE in the stand-alone VS Code. Wpold appreciate any help

ivankravets, 2022-06-22:
Could you provide a remote session to debug this issue? If yes, please contact me at me@ikravets.com. Thanks.

plug-it-in, 2022-06-22:
> contact me at ***@***.*** Thanks.

Would teamviewer be OK?

ivankravets, 2022-06-22:
Yes, please send me ID/Pass. Thanks!

Regards,
Ivan

plug-it-in, 2022-06-22:
Sent to private email

Kind regards

Pete Rawlings

plug-it-in, 2022-06-22:
Ivan fixed . Reinstalled Python for all users

Kind regards

Pete Rawlings

closed 2022-06-23 by ivankravets

ivankravets, 2022-06-23:
Please upgrade the PlatformIO IDE extension to the latest 2.5.0 version. The issue with PlatformIO's pre-built Python has been resolved.

plug-it-in, 2022-10-11:
Superb thanks.

commit e0ea21aa (2022-06-23): Fixed a bug with the handling of package registry mirrors // Resolve #3222