platformio/platformio-vscode-ide #3225: Start PIO Home Server

Potatotatotato, 2022-06-24:
%23 Description of problem
  Leave a comment...

  BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
  - https://github.com/platformio/platformio-vscode-ide/issues%3Fq=is%3Aissue 

  %23 Configuration

  VSCode: 1.65.2
  PIO IDE: v2.5.0
  System: Windows_NT, 10.0.22000, x64

  %23 Exception
  ```
  Error: Error: Traceback (most recent call last):
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  %23 pylint: disable=no-value-for-parameter
    ^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\platformio\cli.py", line 71, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\click\core.py", line 1651, in invoke
    cmd_name, cmd, args = self.resolve_command(ctx, args)
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\click\core.py", line 1698, in resolve_command
    cmd = self.get_command(ctx, cmd_name)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\platformio\cli.py", line 80, in get_command
    module = importlib.import_module(commands[cmd_name])
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\ExpressLRS Configurator\dependencies\windows_amd64\python-3.11.0a4.amd64\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 915, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\platformio\home\cli.py", line 20, in <module>
    from platformio.home.run import run_server
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\platformio\home\run.py", line 35, in <module>
    from platformio.home.rpc.handlers.project import ProjectRPC
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\platformio\home\rpc\handlers\project.py", line 29, in <module>
    from platformio.project.generator import ProjectGenerator
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\platformio\project\generator.py", line 19, in <module>
    import bottle
    ^^^^^^^^^^^^^
  File "C:\Users\28715\.platformio\penv\Lib\site-packages\bottle.py", line 44, in <module>
    from inspect import getargspec
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ImportError: cannot import name 'getargspec' from 'inspect' (C:\Program Files\ExpressLRS Configurator\dependencies\windows_amd64\python-3.11.0a4.amd64\Lib\inspect.py)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

    at cmd (c:\Users\28715\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\home.js:195:11)
    at ChildProcess.c (c:\Users\28715\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\platformio-node-helpers\dist\webpack:\platformio-node-helpers\src\proc.js:177:7)
    at ChildProcess.emit (events.js:315:20)
    at ChildProcess.EventEmitter.emit (domain.js:467:12)
    at ChildProcess.cp.emit (c:\Users\28715\.vscode\extensions\platformio.platformio-ide-2.5.0\node_modules\cross-spawn\lib\enoent.js:34:29)
    at maybeClose (internal/child_process.js:1048:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:288:5)
  ```

ivankravets, 2022-06-27:
Could you remove `C:\Users\28715\.platformio\penv` folder and restart VSCode. Does it work now?

sakitume, 2022-07-05:
FYI I've experienced the same issue as reported here. I searched this issues section and found this (and other identical issues...but this was the oldest).

I took your suggestion @ivankravets to delete the `penv` folder and then restart VSCode. However I had to also be sure to open a PlatformIO project folder so that pio would recognize it had to re-install (parts?) of itself and I noticed it re-generated the `penv` folder and now my installation seems to work fine.

closed 2022-07-06 by ivankravets